dsmrreader/dsmr-reader #1922: 🙋 Probleem met gasmeter?

AbelLykens, 2023-11-02:
### Description

Ben verhuisd, heb DSMR-versie teruggezet naar `2/3`, toen begonnen energiestats weer te werken. Maar ik krijg geen gasmetingen door.

Is er iets mis aan mijn kant?

```
/XMX5LGF0010460945412

1-3:0.2.8(50)
0-0:1.0.0(000101000000W)
0-0:96.1.1(4530303637303036303934353431323233)
1-0:1.8.1(000103.962*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.435*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00004)
0-0:96.7.9(00004)
1-0:99.97.0(3)(0-0:96.7.19)(000101000000W)(0000002216*s)(000101000000W)(0000005784*s)(000101000000W)(0000000545*s)
1-0:32.32.0(00002)
1-0:52.32.0(00001)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(239.5*V)
1-0:52.7.0(236.0*V)
1-0:72.7.0(237.3*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.008*kW)
1-0:41.7.0(00.037*kW)
1-0:61.7.0(00.386*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
!51D8
```

### DSMR-reader version

v5.10.3 

### DSMR-reader platform

None

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.3 / v3.11.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "3"

DATA
    Telegrams total (est.)                                                                             374634
    Consumption records electricity / gas (est.)                                                80515 / 53361
```

dennissiemensma, 2023-11-04:
Het lijkt erop dat je gasmeter niet (goed) gekoppeld is:

```
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
```

Ter vergelijking mijn meter (al is die v4):

```
0-1:24.1.0(003)
0-1:96.1.0(47303030323334<lang getal>)
0-1:24.2.1(231104220000W)(04518.204*m3)
```

AbelLykens, 2023-11-05:
Thanks zal morgen Liander ff bellen dan!

closed 2023-11-05 by AbelLykens