dsmrreader/dsmr-reader #1925: Notities maken per maand/jaar

niekniek89, 2023-11-20:
### Description

Goedemorgen,

Wellicht niet heel spannend (en geen idee of het iets toevoegt).
Ik zie nu dat je per dag een notitie kan maken in het archief van DSMR Reader.
Is het ook mogelijk om dit beschikbaar te maken voor de maanden/jaren?

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-11-20:
Bedankt voor je suggestie. Ik denk dat het relatief veel werk is, gezien de notities ook qua opslag aan een datum hangen en niet perse aan een maand of periode. 
Het is opzich een leuk idee, alleen gezien alle andere prioriteiten denk ik niet dat het haalbaar is.

niekniek89, 2023-11-21:
Dank voor je antwoord.
Ik kan zelf niet bepalen wat wel/niet veel werk is, maar begrijp de keuze zeker wel.
Mocht het ooit eens lukken, zou het inderdaad een leuke toevoeging zijn, maar niet noodzakelijk.

dennissiemensma, 2024-02-23:
Ik sluit deze omdat ik weet dat ik er uiteindelijk niet aan toe ga komen.

closed 2024-02-23 by dennissiemensma