dsmrreader/dsmr-reader #1927: nieuwe installatie geeft fouten "psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above"

braham2019, 2023-11-20:
### Description

wil mijn bestaande installatie overzetten naar Debian Bookworm. Heb andere Pi4 genomen en het installatiedocument doorlopen. 

Bij de stap Importing a full backup ./deploy.sh gaat het mis en krijg ik de melding "psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above"

### DSMR-reader version

new installation

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
--- You are currently running version: 
5.10.3

 --- Checking Python version.
Running Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether '.venv' VirtualEnv is activated.
     [i] ----- Activating '.venv' VirtualEnv...

 --- Checking (minimum) Python version.
Running Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]

 --- Checking & synchronizing base requirements for changes.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: attrs==22.2.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (22.2.0)
Requirement already satisfied: coreapi==2.3.3 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (1.7)
Requirement already satisfied: django==3.2.16 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.2.16)
Requirement already satisfied: django-admin-rangefilter==0.8.8 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.8.8)
Requirement already satisfied: django-admin-sortable==2.3 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.3)
Requirement already satisfied: django-colorfield==0.7.3 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (0.7.3)
Requirement already satisfied: django-debug-toolbar==3.8.1 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.8.1)
Requirement already satisfied: django-filter==21.1 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (21.1)
Requirement already satisfied: django-mathfilters==1.0.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.0.0)
Requirement already satisfied: djangorestframework==3.14.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (3.14.0)
Requirement already satisfied: django-solo==2.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (2.0.0)
Requirement already satisfied: dropbox==11.36.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (11.36.0)
Requirement already satisfied: gunicorn==20.1.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (20.1.0)
Requirement already satisfied: influxdb-client==1.35.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (1.35.0)
Requirement already satisfied: paho-mqtt==1.6.1 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (1.6.1)
Requirement already satisfied: psycopg2-binary==2.9.5 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (2.9.5)
Requirement already satisfied: pyserial==3.5 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (3.5)
Requirement already satisfied: pyserial-asyncio==0.6 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (0.6)
Requirement already satisfied: python-dateutil==2.8.2 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (2.8.2)
Requirement already satisfied: python-decouple==3.7 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 21)) (3.7)
Requirement already satisfied: pytz==2022.7.1 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 22)) (2022.7.1)
Requirement already satisfied: pyyaml==6.0 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 23)) (6.0)
Requirement already satisfied: requests==2.28.2 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 24)) (2.28.2)
Requirement already satisfied: urllib3==1.26.14 in ./.venv/lib/python3.11/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 25)) (1.26.14)
Requirement already satisfied: coreschema in ./.venv/lib/python3.11/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (0.0.4)
Requirement already satisfied: itypes in ./.venv/lib/python3.11/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.2.0)
Requirement already satisfied: uritemplate in ./.venv/lib/python3.11/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (4.1.1)
Requirement already satisfied: asgiref<4,>=3.3.2 in ./.venv/lib/python3.11/site-packages (from django==3.2.16->-r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.7.2)
Requirement already satisfied: sqlparse>=0.2.2 in ./.venv/lib/python3.11/site-packages (from django==3.2.16->-r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.4.4)
[... 9 lines cut ...]
Requirement already satisfied: ply>=3.4 in ./.venv/lib/python3.11/site-packages (from stone>=2.*->dropbox==11.36.0->-r dsmrreader/provisioning/requirements/base.txt (line 13)) (3.11)
Requirement already satisfied: jinja2 in ./.venv/lib/python3.11/site-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (3.1.2)
Requirement already satisfied: MarkupSafe>=2.0 in ./.venv/lib/python3.11/site-packages (from jinja2->coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.1.3)

 --- Applying database migrations.
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/django/utils/asyncio.py", line 33, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/psycopg2/__init__.py", line 122, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
psycopg2.OperationalError: SCRAM authentication requires libpq version 10 or above
```

dennissiemensma, 2023-11-20:
Bedankt voor je melding, zie ook #1624. Daar staan nog wat suggesties in die je kunt proberen.

dennissiemensma, 2023-12-04:
Bedankt voor je update. 
- Doen de andere twee `dsmr_` processen het wel? 

Je kunt nog proberen om `libpq-dev` te installeren op je OS en het dan de hele virtualenv opnieuw te installeren. Wellicht pakt die dan bij installeren van de database-client de juist versie (meestal kijkt die naar het OS, dus later die package installeren in het OS werkt doorgaan alleen als je de virtualenv dan opnieuw aanmaakt).

```
sudo su - dsmr
deactivate
rm -rf ~/dsmr-reader/.venv/

python3 -m venv ~/dsmr-reader/.venv/

source ~/dsmr-reader/.venv/bin/activate
pip3 install pip --upgrade
pip3 install -r ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
```

joopy, 2023-12-04:
Bedankt voor je reactie! Het was voor de raspberry pi van mijn vader, dus heb vandaag 6 keer een nieuw image van raspbian geinstalleerd en verschillende stappen geprobeerd. Uiteindelijk werkt nu de docker methode.

closed 2023-12-12 by dennissiemensma