dsmrreader/dsmr-reader #1928: 🙋Ik heb mijn server naar FreeBSD 14 geupgrade backend: ok / datalogger: ok / Webinterface NOK

heyudude, 2023-11-22:
### Description

Ik ben bezig met de migratie naar een nieuwe server op FreeBSD 14.
Installatie is gelukt, supervisor configuratie lijkt goed.
Backend en datalogger draaien inmiddels.

Echter de webinterface geeft de volgende error.
ModuleNotFoundError: No module named 'colorfield'

De .venv installatie verliep prima: django-colorfield is geinstalleerd zie ik
en in /usr/local/www/dsmread/static is een subdirectory colorfield te zien.

NB geen P1 gekoppeld....

Geen idee of de stap naar release 14 te snel is (net uit) of dat ik iets gemist heb?
Alle tips zijn welkom

### DSMR-reader version

DSMR-reader: v5.10.4

### DSMR-reader platform

Other

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.4 / v3.9.18 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                 -1
    Consumption records electricity / gas (est.)                                                      -1 / -1

UNRESOLVED ISSUES
    No day statistics found                                                                               now
    Waiting for the first reading ever                                                                    now
```

dennissiemensma, 2023-11-24:
Bedankt voor je melding. Het is een wat vreemde fout, gezien alle drie de processen gewoon de code van DSMR-reader laden (en er geen verschil zou moeten zijn). Ik kan me niet herinneren dat ik dat eerder heb gezien. Meestal gaan ze dan allemaal stuk op dezelfde fout.

Je kunt nog even alle Python cache file weggooien en daarna herstarten. Maar dat is puur een probeersel:

```
sudo su - dsmr
find . -type f -iname "*.pyc"

# Als je alleen .PYC files ziet, verwijder ze met -delete
find . -type f -iname "*.pyc" -delete

./reload.sh
```

closed 2023-12-12 by dennissiemensma

heyudude, 2023-12-13:
Sorry voor de late reactie. Dank voor de tips.
Alles nog eens goed doorgelopen. Denk een stap gemist. En nu werkt het naar behoren._