dsmrreader/dsmr-reader #1929: Ondersteuning Honewizard P1 meter

petermdevries, 2023-11-30:
### Feature

Is het een idee om de HomeWizard P1 meter als datalogger te kunnen gebruiken? Deze heeft de mogelijkheid om een lokale API beschikbaar te stellen waarmee o.a. de laatste telegram opgehaald kan worden.

Ik gebruik nu zelf nog een raspberry pi met kabel, maar overweeg de HomeWizard P1 meter. DSMR reader zou ik er dan ook wel naast willen kunnen blijven gebruiken.

https://api-documentation.homewizard.com/

dennissiemensma, 2023-11-30:
Bedankt voor je suggestie, zie ook:
- #1327
- #1664

De TL;DR is dat het wat lastig is om alle andere apparaten te ondersteunen. Als ik er eentje toevoeg, moet ik ze allemaal ondersteunen en dat is helaas geen doen.

closed 2023-11-30 by dennissiemensma

petermdevries, 2023-12-02:
Dank voor de tips. 

Geïnspireerd door #1327 heb ik het voorbeeld plugin 5 (https://dsmr-reader.readthedocs.io/nl/latest/reference/plugins.html#example-5-read-telegrams-using-dsmrloggerws-api) gebruikt. Door de DSMRLOGGERWS_ENDPOINT simpelweg aan de te passen naar 'http://<hwp1meter_address>/api/v1/telegram' werkt het.

Waar in het verleden de CRC code leek te ontbreken is deze nu/tegenwoordig gewoon aanwezig.

Is er een manier om de bestaande datalogger (momenteel via netwerk socket) dan uit te zetten? Ik zie in de handleiding een stukje staan over het verwijderen van de /etc/supervisor/conf.d/dsmr_datalogger.conf. Zou dat de juiste manier zijn voor mijn situatie?

migube, 2024-01-12:
Hi [petermdevries] and [dennissiemensma], 
dus als ik het goed begrijp zou de Homewizard optie moeten werken, via de plugin ?
is er een manier om de plugin op een home assistant versie van dsrm reader aan te maken? ik vond geen opties hiervoor :(
Kan de pluging anders in de volgende versie ingebouwd worden, met in de GUI de variable voor de ENDPOINT?

dank u!
migube

dennissiemensma, 2024-01-15:
@migube er is geen officiele ondersteuning. Er zijn teveel tooltjes en soortgelijke apparaten om ze allemaal (of een deel) te ondersteunen.