dsmrreader/dsmr-reader #1932: 🙋Dagstatistieken lopen achter

supershadow, 2023-12-05:
### Description

de dagstatistieken loop al 3 weken achter en word niet meer bij gewerkt.
nu was ik in de instellingen aan na kijken en zag dat Influxdb-Integratie is uitgeschakeld.
nu heb ik geprobeerd weer in te schakelen maak dan vraag die naar influxDB-organization en influxDB API-token maar ik heb geen idee wat dit moet zijn. enige idee waar ik dit kan terug kan vinden ?

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-12-05:
Je kunt het beste de logs induiken: https://github.com/dsmrreader/dsmr-reader/issues/1230#issuecomment-742636653

Voor InfluxDB geldt hetzelfde. Let wel op dat het influxdb versie 2 is.

supershadow, 2023-12-05:
ik heb exact zelfde probleem als 
Dagstatistieken worden niet gegenereerd na uitschakelen gas #1197
maar nog geen idee hoe ik dat kan oplossen

supershadow, 2023-12-07:
heb het al gevonden moest gas in de settings uitvinken