dsmrreader/dsmr-reader #1935: 🙋504 Gateway Time-out

rustig136, 2023-12-11:
### Description

DSMR, docker installatie.
Ik vermoed dat de Raspberry het niet meer kan bijhouden of dat de database te groot is geworden.
Wat ik al geprobeerd heb: docker exec -ti dsmr bash -c '/app/cleandb.sh' en de DSMRREADER_REMOTE_DATALOGGER_SLEEP=5 toegevoegd aan de docker-compose.yaml
Dit alles heeft het nog niet opgelost. In de logging zie ik onderstaande:

Wat kan ik nog doen? Alvast bedankt voor je support!
Mike

### DSMR-reader version

5

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
today at 15:35  [cont-init.d] 20-set-app-defaults: exited 0.
today at 15:35  [cont-init.d] done.
today at 15:35  [services.d] starting services
today at 15:35  [services.d] done.
today at 15:35  Starting DSMR Reader - backend...
today at 15:35  Starting DSMR Reader - webinterface...
today at 15:35  Starting DSMR Reader - datalogger...
today at 15:35  Starting DSMR Reader - nginx...
today at 15:35  [2023-12-11 14:35:57 +0000] [478] [INFO] Starting gunicorn 20.1.0
today at 15:35  [2023-12-11 14:35:57 +0000] [478] [INFO] Listening at: unix:/tmp/gunicorn--dsmr_webinterface.socket (478)
today at 15:35  [2023-12-11 14:35:57 +0000] [478] [INFO] Using worker: sync
today at 15:35  [2023-12-11 14:35:57 +0000] [502] [INFO] Booting worker with pid: 502
today at 15:36  Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
today at 15:36  Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
today at 15:37  2023/12/11 14:37:03 [error] 498#498: *1 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:37  127.0.0.1 - - [11/Dec/2023:14:37:03 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:37  [2023-12-11 14:37:19 +0000] [478] [CRITICAL] WORKER TIMEOUT (pid:502)
today at 15:37  [2023-12-11 14:37:20 +0000] [478] [WARNING] Worker with pid 502 was terminated due to signal 9
today at 15:37  [2023-12-11 14:37:21 +0000] [538] [INFO] Booting worker with pid: 538
today at 15:37  2023/12/11 14:37:22 [error] 499#499: *3 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:37  127.0.0.1 - - [11/Dec/2023:14:37:22 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:37  2023/12/11 14:37:40 [error] 500#500: *5 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:37  127.0.0.1 - - [11/Dec/2023:14:37:40 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:38  2023/12/11 14:38:00 [error] 501#501: *7 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:38  127.0.0.1 - - [11/Dec/2023:14:38:00 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:38  2023/12/11 14:38:18 [error] 498#498: *9 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:38  127.0.0.1 - - [11/Dec/2023:14:38:18 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:38  2023/12/11 14:38:37 [error] 498#498: *11 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:38  127.0.0.1 - - [11/Dec/2023:14:38:37 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:38  127.0.0.1 - - [11/Dec/2023:14:38:57 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:38  2023/12/11 14:38:57 [error] 498#498: *13 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:39  127.0.0.1 - - [11/Dec/2023:14:39:14 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:39  2023/12/11 14:39:14 [error] 498#498: *15 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:39  [2023-12-11 14:39:16 +0000] [478] [CRITICAL] WORKER TIMEOUT (pid:538)
today at 15:39  [2023-12-11 14:39:18 +0000] [478] [WARNING] Worker with pid 538 was terminated due to signal 9
today at 15:39  [2023-12-11 14:39:18 +0000] [580] [INFO] Booting worker with pid: 580
today at 15:39  2023/12/11 14:39:33 [error] 498#498: *17 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:39  127.0.0.1 - - [11/Dec/2023:14:39:33 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:39  2023/12/11 14:39:55 [error] 498#498: *19 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:39  127.0.0.1 - - [11/Dec/2023:14:39:55 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:40  2023/12/11 14:40:14 [error] 498#498: *21 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:40  127.0.0.1 - - [11/Dec/2023:14:40:14 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:40  2023/12/11 14:40:33 [error] 498#498: *23 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:40  127.0.0.1 - - [11/Dec/2023:14:40:33 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
today at 15:40  2023/12/11 14:40:53 [error] 498#498: *25 upstream timed out (110: Operation timed out) while reading response header from upstream, client: 127.0.0.1, server: _, request: "GET /about HTTP/1.1", upstream: "http://unix:///tmp/gunicorn--dsmr_webinterface.socket/about", host: "127.0.0.1"
today at 15:40  127.0.0.1 - - [11/Dec/2023:14:40:53 +0000] "GET /about HTTP/1.1" 504 160 "-" "curl/8.4.0" "-"
```

dennissiemensma, 2023-12-11:
Heb je ook al de verbose variant geprobeerd? Zie je daar wel output?
https://github.com/xirixiz/dsmr-reader-docker#dsmr-reader---database-cleanupvacuum

Verder kun je nog kijken of in je de docker-container `./manage.py dsmr_debuginfo` kan uitvoeren. Die output helpt ook bij checken of het de hoeveelheid data is.

En anders zul je zelf de database in moeten en wat counts doen. 
```
sudo su - postgres
psql dsmrreader
select count(id) from dsmr_datalogger_dsmrreading;
```

Commands kunnen wat anders zijn in een DB-container in Docker.

rustig136, 2023-12-18:
Dank je wel voor je snelle reactie. Ik heb in de tussentijd het probleem gevonden. Het was een defecte SD kaart. De impact was zo groot dat DSMR niet meer kon functioneren. Ik ben opnieuw begonnen met een nieuwe V5 installatie. Ben nu aan het kijken of ik de Docker container nog kan benaderen om de backup eraf te halen.

closed 2024-01-11 by dennissiemensma