dsmrreader/dsmr-reader #1944: 🚨Geen MQTT berichten voor dsmr/current-period op nieuwjaarsdag

RichieB2B, 2024-01-02:
### Description

Normaliter verschijnt in het MQTT topic `dsmr/current-period` elke minuut een update. Echter is er in dit topic op nieuwjaarsdag (1 januari 2024) geen enkel bericht geplaatst in dit topic. Pas op 2 januari 2024 om 00:02 uur verscheen daar het eerste bericht van het jaar.

Ik vermoed dat dit te maken heeft met #1811 die wacht op de refactoring voor #1770

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-01-02:
Bedankt voor je melding. Gezien het slechts 1x per jaar voor komt, geef ik deze wat lage prio

RichieB2B, 2025-01-01:
Vandaag trad dit probleem (uiteraard) weer op. Dit keer had ik de tijd om het te troubleshooten, zie #2045

dennissiemensma, 2026-08-18:
Onderdeel van [v6.2](https://github.com/dsmrreader/dsmr-reader/releases/tag/v6.2.0), nieuwe container build volgt komend uur