dsmrreader/dsmr-reader #1946: 🙋Nieuwjaars error. No port configured, configure a UI port in the Network options

dennispracht, 2024-01-03:
### Description

Goededag en nog een gelukkig nieuwjaar.

Waar ze bang voor waren toen de datum van 1999 naar 2000 ging. Daar heb ik nu last van. Nieuwe jaar is begonnen en mijn dsmr werkt niet meer.

Ik zit met het volgende:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 01-set-hassio-vars: executing... 
Home Assistant DSMR Reader Add-on - Release: 1.10.2
Home Assistant DSMR Reader Add-on - Update check failed
Home Assistant DSMR Reader Add-on - Configuring addon settings
Home Assistant DSMR Reader Add-on - No port configured, configure a UI port in the Network options
[cont-init.d] 01-set-hassio-vars: exited 1.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

Ik kan echt nergens vinden wat dat configure a UI port betekend. Het heeft het altijd gedaan. Port is het zelfde in proxmox. tty0

### DSMR-reader version

5

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-01-04:
Bedankt voor je melding. Dit project gaat puur over DSMR-reader en niet de integratie met HA.

Kan het zijn dat je deze hebt geinstalleerd?
- https://github.com/sanderdw/hassio-addons

Wellicht kan Sander je daar op weg helpen.

dennispracht, 2024-01-07:
Bedankt voor je reactie. 

Als ik naar de ip ga van dsmr en dan kom ik netjes op de webpagina terecht. Zoals al 1 jaar staat mijn usb op tty0. Als ik mijn computer opnieuw opstart en alles is weer online. Dan zie ik op de webpagina van dsmr dat die stroom aan meten is. Maar dit doet die maar voor 10 seconden. Dan stopt die ermee. 

Dsmr doet dus iets niet meer goed. Waardoor de integratie van ha het ook niet doet.

Die metadata heb ik al een jaar samen draaien in ha. Dus dat lijkt me sterk dat het daarom fout gaat. Zonder ha doet dsmr ook raar.

dennissiemensma, 2024-01-08:
Dat DSMR-reader stopt/herstart wil niet zeggen dat het in DSMR-reader zit, want ik zie ook `01-set-hassio-vars: exited 1.`, een `exit 1` op een hassio script. DSMR-reader kun je op 3 of 4 verschillende manieren draaien, dit is er eentje van.

```
Home Assistant DSMR Reader Add-on - No port configured, configure a UI port in the Network options
[cont-init.d] 01-set-hassio-vars: exited 1.
```

Het is wellicht wat lastig, maar ik kan je oprecht niet helpen met HA issues of die van de add-on. Alleen foutmeldingen van DSMR-reader zelf. Wellicht dat @sanderdw nog een suggestie voor je heeft voro debuggen.

En anders zul je de logs in moeten van het falende proces (als je dat al ergens kan zien). Daar staat meestal wel een reden.

dennissiemensma, 2024-01-08:
https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/logfiles.html

sanderdw, 2024-01-09:
Zoals @dennissiemensma al zegt heeft dit te maken met the addon configuratie en niet de Dsmr Reader software zelf. Zie ook mijn reactie onder https://community.home-assistant.io/t/dsmr-reader-add-on-for-home-assistant/279087/402?u=sanderdw.

Zou dus voornamelijk even die thread gebruiken om te onderzoeken wat het probleem is.

dennispracht, 2024-01-10:
Het werkt weer. Weet echt niet wat het was. Maar heb dsmr, metabase en timescale opnieuw geïnstalleerd en opnieuw alle gegevens erin gezet. Nu werkt het weer. Bedankt voor je reacties.

dennissiemensma, 2024-01-10:
Fijn om te horen, dank voor je update