dsmrreader/dsmr-reader #1947: 🚨 Mail versturen met SSL of TLS geeft: SMTP_SSL.__init__() got an unexpected keyword argument 'keyfile'

casakampa, 2024-01-06:
### Description

Het versturen van een testmail met versleuteling (SSL of TLS) geeft deze melding in het adminpanel:

`Failed to send test email: SMTP.starttls() got an unexpected keyword argument 'keyfile'`

Zowel de email-, server- als inloggegevens zijn getest en correct.

Zelf zoeken heeft het volgende opgeleverd:

https://docs.python.org/3/library/smtplib.html#smtplib.SMTP_SSL

`Changed in version 3.12: The deprecated keyfile and certfile parameters have been removed.`

### DSMR-reader version

5.10.4

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2024-01-08:
Bedankt voor je melding. Welke Python-versie gebruikt die image?

Want Python `3.12` wordt **niet** ondersteund door DSMR-reader. Alleen Python `3.7` tot `3.11`.

casakampa, 2024-01-10:
> Bedankt voor je melding. Welke Python-versie gebruikt die image?
> 
> Want Python `3.12` wordt **niet** ondersteund door DSMR-reader. Alleen Python `3.7` tot `3.11`.

Na onderzoek blijkt de gebruikte container Python versie 3.12 te draaien.
Dan is duidelijk waar de oorzaak vandaan komt.

closed 2024-01-10 by casakampa

dennissiemensma, 2024-01-11:
Nog een correctie op mezelf hierboven, Python 3.11 wordt ook niet ondersteund omdat Djang0 3.x (in DSMR-reader v5) die ook niet ondersteunt.

Python ondersteuning: `3.7, 3.8, 3.9, 3.10`

Bij de volgende major update van DSMR-reader zal het weer opschuiven. Dat kan zomaar Django 5.x worden met Python 3.10, 3.11, 3.12. Maar dat komt tzt wel in de release notes