dsmrreader/dsmr-reader #1948: Dropbox integratie probleem🚨

ridderr, 2024-01-08:
### Description

Ik heb de laatste dagen onderstaande foutmelding. Na het aanmaken van een nieuwe token komt deze foutmelding weer terug na enige uren.

`Jan. 8, 2024: Unable to authenticate with Dropbox, removing credentials. Error: InternalServerError('6e1e4798caeb43a5865205fd9d0619dc', 504, '<!DOCTYPE html>\n<html>\n<head><meta http-equiv="Content-Type" content="text/html; charset=utf-8">\n<meta name="viewport" content="width=device-width, initial-scale=1" />\n<title>Dropbox - 504</title>\n<link href="https://cfl.dropboxstatic.com/static/metaserver/static/css/error.css" rel="stylesheet" type="text/css"/>\n<link rel="shortcut icon" href="https://cfl.dropboxstatic.com/static/images/favicon.ico"/>\n</head>\n<body>\n<div class="figure">\n[image]\n</div>\n<div id="errorbox">\n<h1>Error (504)</h1>Something went wrong. Don\'t worry, your files are still safe and the Dropbox team has been notified. Check out our <a href="https://status.dropbox.com">Status Page</a> to see if there is a known incident, our <a href="https://www.dropbox.com/help">Help Center</a> and <a href="https://forums.dropbox.com">forums</a> for help, or head back to <a href="https://www.dropbox.com/home">home</a>.\n</div>\n\n</body>\n</html>\n')`

### DSMR-reader version

v5.10.4

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-01-09:
Bedankt voor je melding. Helaas zit dit in dit geval aan de kant van Dropbox. Het is een onverwachte server fout aan hun zijde, want normaal gesproken komt er een specifieke fout uit met de exacte oorzaak. Helaas staat dat er nu ook niet bij, dus ik heb geen oplossing voor je.

ridderr, 2024-01-09:
Hoi Dennis,
Daar was ik al bang voor. Het aanvragen van het token gaat goed maar na enige tijd (tijdens de backup) dan gebeurd er ergens wat en is het token ongeldig gemaakt. 
Maar even wachten tot de dropbox jongens/meisjes het probleem oplossen.

closed 2024-01-09 by ridderr