dsmrreader/dsmr-reader #1955: Gas-verbruik wordt niet geïmporteerd

MathijsG, 2024-01-24:
### Communicatietaal / Language

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### DSMR-reader version

v5.10.4

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
Ik heb het Python-script gedraaid uit deze thread: #520 en #1892

Deze heb ik gedraaid, en hierna heb ik keurig de energie-data vanaf 2016 in mijn systeem. Echter zijn mijn gasmeter-standen compleet leeggebleven. Ik zie ze wel gevuld worden in de CSV-file, dus weet niet helemaal waar ik moet beginnen te debuggen.

De datastructuur uit het script is:

`` data={
       'electricity_currently_delivered': 0.000,
       'electricity_currently_returned': 0.000,
       'electricity_delivered_1': Del1,
       'electricity_returned_1': Ret1,
       'electricity_delivered_2': Del2,
       'electricity_returned_2': Ret2,
       'extra_device_timestamp': timestamp,
       'extra_device_delivered': Gas,
       'timestamp': timestamp
      }
    )``

Is daar wellicht iets mis mee? Deze structuur klopt wel met de CSV-kolommen

Ook zie ik in de logs van het script bijvoorbeeld:

``2021-03-01 6043749 0 6870851 0 1534082
Created: {'id': 23647, 'timestamp': '2021-02-28T23:00:00+01:00', 'electricity_delivered_1': '6043.749', 'electricity_returned_1': '0.000', 'electricity_delivered_2': '6870.851', 'electricity_returned_2': '0.000', 'electricity_currently_delivered': '0.000', 'electricity_currently_returned': '0.000', 'phase_currently_delivered_l1': None, 'phase_currently_delivered_l2': None, 'phase_currently_delivered_l3': None, 'extra_device_timestamp': '2021-02-28T23:00:00+01:00', 'extra_device_delivered': '1534.082', 'phase_currently_returned_l1': None, 'phase_currently_returned_l2': None, 'phase_currently_returned_l3': None, 'phase_voltage_l1': None, 'phase_voltage_l2': None, 'phase_voltage_l3': None, 'phase_power_current_l1': None, 'phase_power_current_l2': None, 'phase_power_current_l3': None}
``

Waarbij met 'extra_device_delivered': '1534.082' het gasverbruik lijkt te worden aangegeven.

Ik zie hem ook wel onderaan als verbruik staan (de datum lijkt 1 dag verschoven, dat is een ander issue dat ik even moet fixen ergens), maar niet bij verbruik zelf.
```

MathijsG, 2024-01-24:
[image]

Zie hier wat ik bedoel. Moet ik daar nog weer iets anders voor draaien na importeren? Ik heb ./manage.py dsmr_stats_reconstruct_missing_day_statistics al gedraaid

dennissiemensma, 2024-01-26:
Bedankt voor je melding. Ik heb er geen directe verklaring voor. Dan zul je de DEBUG logs van het backend-proces moeten bekijken of je daar iets ziet over gas:

- https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

dennissiemensma, 2024-01-26:
Check verder even of je meterstanden wel in de goede tijdzones meekomen, gezien ze een uur lijken af te wijken in je screenshot. Ik kan me heugen dat dat iets was met importscripts.

Staat overigens los van je gas-probleem.

closed 2024-02-06 by dennissiemensma