dsmrreader/dsmr-reader #1965: SyntaxWarning: invalid escape sequence '\d'

jicho, 2024-02-23:
### Communicatietaal / Language

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Zojuist heb ik de laatste versie van DSMR-reader geïnstalleerd via de update van xirixiz/dsmr-reader-docker.

Alles draait soepel, maar bij het opstarten kreeg ik deze melding in de docker CLI logs:
```
dsmr-reader  | /app/dsmr_parser/parsers.py:41: SyntaxWarning: invalid escape sequence '\d'
dsmr-reader  |   """
```

Dit is niet iets wat me afkomstig lijkt te zijn door de container, maar meer vanuit de Python code.

Wou het even melden, heb geen idee of de melding een probleem gaat veroorzaken in de juiste werking. Alles lijkt in ieder geval zijn ding te doen.

Of ligt het aan de Python versie in de container?
```
root@dsmr_reader_docker:/app# python -V
Python 3.12.2
```

### DSMR-reader version

v5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-02-23:
Bedankt voor je melding. Python 3.12 wordt nog *niet* ondersteund door DSMR-reader, dus grote kans dat dat het is!

jicho, 2024-02-23:
Dank voor de snelle reactie.
'k Zie hier idd wel iets https://docs.python.org/3/whatsnew/3.12.html rondom de [SyntaxWarning](https://docs.python.org/3/library/exceptions.html#SyntaxWarning)

Eigenlijk wel apart dat er "gepiept" wordt over een regex in de documentatie van een functie in parsers.py. Zou je eigenlijk niet verwachten... Toch?

dennissiemensma, 2024-02-23:
Ja grote kans dat het Python 3.12 is, want Python-versies zijn niet perse compatible met vorige versies. 

Het staat wel op de planning om te upgraden overigens, maar dat is een grote klus omdat ik ook het onderliggende framework framework moet upgraden. En daardoor moet ik ook weer oude database-versies droppen.

- #1763 
- #1753
- #1594

Dus grote kans dat het tzt een grote bump gaat worden, waarbij 3 Python-versies gedropt worden en een hoop oude databases.

En omdat het upgraden zoveel gedoe is voor veel gebruikers, wordt dat vermoedelijk ook het moment dat ik niet meer "native" DSMR-reader ondersteun, maar alleen nog via Docker.

closed 2024-02-23 by dennissiemensma