dsmrreader/dsmr-reader #1971: deploy.sh: Upgrade issue bij  --- Applying database migrations.

heyudude, 2024-03-02:
### Communicatietaal / Language

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

deploy.sh: Upgrade issue bij  --- Applying database migrations.

ImportError: Shared object "libimagequant.so.0.0.4" not found, required by "_imaging.cpython-39.so"
>>>>> [!] Executing database migrations failed again! <<<<<

Ik heb libimagequant geinstalleer al is die niet genoemd als dependancy.
Oplossing? Upgrade Python of? 

### DSMR-reader version

 v5.10.4

### DSMR-reader platform

Native (e.g. manual installation)

heyudude, 2024-03-02:
Opgelost door:   ln -sf /usr/local/lib/libimagequant.so.0.4.0 /usr/local/lib/libimagequant.so.0.0.4

Binnenkort maar eens naar Python3.11 upgraden ;)

dennissiemensma, 2024-03-02:
Bedankt voor je melding en oplossing! 

Als je een keertje Python gaat upgraden, kan ik je aanraden om in plaats daarvan over te stappen op de Docker-versie van Xirixiz. Dat maakt updates (vooral toekomstige) stukken makkelijker. Plus dat ik uiteindelijk ook alleen die variant ga ondersteunen.

closed 2024-03-02 by dennissiemensma