dsmrreader/dsmr-reader #1975: Postgres14.5 en backup lokatie NAS , DSMR op PI probleem nu met backup file.

mupsje, 2024-03-19:
### Communicatietaal / Language

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Wegens verhuizing heb ik mijn DSMR P1 op een PI moeten zetten.

//-------------------------------------------------------------------------------------//
Oude situatie: 
//-------------------------------------------------------------------------------------//

QNAP NAS, Container docker:
100% werkend!

dmsrdb Postgres op docker 192.168.86.234
dsmr-reader        op docker 192.168.86.233
    volumes:
      - /share/docker/dsmr_backups:/app/backups
gemiddelde backup file .sql.gz +/- 15MB

//-------------------------------------------------------------------------------------//
Nieuwe situatie:
//-------------------------------------------------------------------------------------//

PI - volgens handleiding geinstalleerd behalve stap
1. Database backend (PostgreSQL)
Want die heb ik draaiend op de QNAP
dus heb in in de /home/dsmr/dsmr-reader/.env 

```
DJANGO_DATABASE_ENGINE=django.db.backends.postgresql
DJANGO_DATABASE_HOST=192.168.86.234
DJANGO_DATABASE_PORT=5432
DJANGO_DATABASE_NAME=dsmrreader
DJANGO_DATABASE_USER=dsmrreader
DJANGO_DATABASE_PASSWORD=^%$#^$%$#@^$%#$$#@%$
etc.
```
[image]

Omdat deze op de PI draait heb ik een NFS map gemaakt op de NAS en deze map is bereikbaar voor *.
en in /etc/fstab volgende gezet.
```
NASCURA1:/DSMR          /home/dsmr/dsmr-reader/backupdsmr               nfs     defaults 0 0
```
dsmr maakt netjes de map archive, jaartal, maand.
[image]

[image]

//-------------------------------------------------------------------------------------//

maar nu komt het probleem . . . 

De backup file wordt ook aangemaakt maar blijft steeds 0KB
Ik heb waarschijnlijk iets over het hoofd gezien.
Maar er wordt dus wel een backup gemaakt maar er staat niks in.

Wie kan mij helpen.

Groet Remco

### DSMR-reader version

 v5.11

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
DSMR-READER
    App | Python | Database                                                       v5.11 | v3.9.2 | postgresql
    BE sleep | DL sleep | Retention | Override                                     1.0s | 5.0s | 744h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                             453085
    Consumption records electricity | gas (est.)                                                65474 | 28473
```

### Optional: Smart meter telegram

```shell-script
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(240319092641W)
0-0:96.1.1(4530303438303030303238323139343138)
1-0:1.8.1(007693.765*kWh)
1-0:1.8.2(004972.587*kWh)
1-0:2.8.1(003249.096*kWh)
1-0:2.8.2(007475.059*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.154*kW)
1-0:2.7.0(00.564*kW)
0-0:96.7.21(00287)
0-0:96.7.9(00021)
1-0:99.97.0(10)(0-0:96.7.19)(231020174443S)(0000000223*s)(230429073205S)(0000007245*s)(230428230937S)(0000000694*s)(221121231905W)(0000000334*s)(221119234752W)(0000000411*s)(221021134225S)(0000000179*s)(220917123921S)(0000000343*s)(220804120708S)(0000000800*s)(220218175953W)(0000000242*s)(200305122618W)(0000000377*s)
1-0:32.32.0(00005)
1-0:52.32.0(00009)
1-0:72.32.0(00035)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(231.0*V)
1-0:52.7.0(231.0*V)
1-0:72.7.0(227.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.005*kW)
1-0:41.7.0(00.148*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.564*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303539303033383334393634353139)
0-1:24.2.1(240319092500W)(05144.010*m3)
!D665
```

dennissiemensma, 2024-03-20:
Je kunt het handmatig proberen in de DSMR-reader container met:
```
./manage.py dsmr_backup_create --compact
```

Mogelijk zie je dan een foutmelding met een oorzaak

dennissiemensma, 2024-03-20:
Onderwater is het overigens een `pgdump`, alleen heb ik geen voorbeeld voor je, omdat redelijk wat variabelen in staan.

Wellicht geeft dit je een indicatie:
- https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_backup/services/backup.py#L104-L116

Als de databaseverbinding het zelf wel doet, zou ik het ook nog zoeken in de schrijflocatie op het bestandssyteem. Al zou ik dan eerder geen bestand verwachten, dan een 0 Bytes bestand...

mupsje, 2024-03-21:
> Je kunt het handmatig proberen in de DSMR-reader container met:
> 
> ```
> ./manage.py dsmr_backup_create --compact
> ```
> 
> Mogelijk zie je dan een foutmelding met een oorzaak

```
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'

```

mmmmmm dit is natuurlijk raar.

mupsje, 2024-03-21:
> Je kunt het handmatig proberen in de DSMR-reader container met:
> 
> ```
> ./manage.py dsmr_backup_create --compact
> ```
> 
> Mogelijk zie je dan een foutmelding met een oorzaak

```
(.venv) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py dsmr_backup_create --compact
2024-03-21 08:05:08,104 INFO     backup       create_partial                  182 |  - Creating new partial backup: /home/dsmr/dsmr-reader/backupdsmr/manually/dsmrreader-postgresql-partial-backup-Thursday.sql
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backup/management/commands/dsmr_backup_create.py", line 42, in handle
    backup_file = dsmr_backup.services.backup.create_partial(
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/backup.py", line 194, in create_partial
    backup_process = subprocess.Popen(  # noqa: S603
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pg_dump'
```

inderdaad een pg_dump fout...
Ik ga eens alles opnieuw installeren met de Postgress op de PI (wat ik eigenlijk niet wil, want ik heb een stand alone)

mupsje, 2024-03-21:
> Onderwater is het overigens een `pgdump`, alleen heb ik geen voorbeeld voor je, omdat redelijk wat variabelen in staan.
>
> Wellicht geeft dit je een indicatie:
>
> * [...]

Dank dank,

Inderdaad ik zie nu dat ik handmatig deze moet doen omdat Postgres in een docker draait.
Ik zal me eens erin verdiepen.
Wanneer ik 

```
docker exec -t dsmrdb pg_dumpall -c -U dsmrreader | /bin/gzip > dsmrreader.sql.gz
```

uitvoer dan krijg ik inderdaad de dump als gz file.

dank allen,
ik ga eens verdiepen in deze.

Groet Remco

dennissiemensma, 2024-03-21:
Dank voor de updates.

Je kunt als alternatief ook overwegen om  alleen de PostgreSQL client op de DSMR-reader machine te installeren, meestal heet die iets als `postgresql-client`.

Echter, dat gaat alleen werken als die client-versie exact overeenkomt met de server-versie (DB versie). Dus het kan ook gedoe opleveren.

closed 2024-03-21 by dennissiemensma