dsmrreader/dsmr-reader #1977: Mindergas upload tijd randomiseren ?

mkruiver, 2024-03-23:
### Communicatietaal / Language

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Misschien kan het upload tijdstip random tussen 00 en 01 uur worden gemaakt?

Ref deze mail:

```
Beste MinderGas deelnemer,

Je maakt gebruik van de Meterstand API van MinderGas.nl om dagelijks je meterstand te uploaden.

Op dit moment posten veel gebruikers hun meterstand om precies 0.00 uur naar MinderGas (of vlak ervoor of vlak erna). De MinderGas.nl website is echter niet ontworpen om vele honderden verzoeken binnen een minuut te verwerken. Deze piekbelasting kan leiden tot response code 503 'Server unavailable'. De meterstand wordt in dat geval niet goed opgeslagen.

Indien je dit wilt voorkomen en je meterstand betrouwbaar wilt uploaden, doe dan de upload op een willekeurig gekozen moment tussen 0.05 en 1.00 uur (of op een ander tijdstip gedurende de dag). Je zult hiervoor je script moeten aanpassen.

Zie ook de API documentatie.

Met vriendelijke groeten, David MinderGas.nl
```

### DSMR-reader version

5

### DSMR-reader platform

Other (e.g. Home Assistant Plugin)

dajappie, 2024-03-23:
Got the same email, somewhere between 0:05 and 01:00 is requested.

dennissiemensma, 2024-03-23:
Bedankt voor je melding. Dit is al eerder aangepast. DSMR-reader uploadt de gasmeting pas later in de nacht, willekeurig tussen 3 en 6 uur. Ik vermoed dat de mail naar alle API gebruikers is gestuurd.

mkruiver, 2024-03-23:
Ok, Bedankt voor de toelichting :)

closed 2024-03-23 by mkruiver