dsmrreader/dsmr-reader #1994: OperationalError: database is locked

mjholtkamp, 2024-05-08:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Hoi! Ik weet niet zeker of dit een bug is of dat ik misschien iets verkeerd doe.

Ik krijg regelmatig de melding: `OperationalError: database is locked`. Voordat ik dit kreeg merkte ik dat de webinterface ook trager was dan voorheen.

Ik gebruik *sqlite* als backend in een docker container.

Ik heb de volgende dingen al gecontroleerd:
* Data retention staat op `Clean up most source data after one week (RECOMMENDED)` (had ik eerder op 1 maand)
* Datalogger process sleep heb ik van 5.0 naar 10.0 gezet.
* Ik heb de database gevacuumed (direct na aanpassing van bovenstaande settings, en enkele dagen erna nog een paar keer). De databasegrootte ging van 126MiB naar 39MiB, maar ik heb nog steeds het probleem.

Ook zag ik eerder al dat er meer IO dan ik verwachtte was naar mijn DSMR lvm logical volume:
[image]

(De drop in IO/s gebeurde rond dezelfde tijd dat ik de settings heb aangepast).

Als ik met iwatch kijk naar mijn DSMR volume zie ik minstens 1x/sec, soms meerdere keren dat er activiteit is op de database:
```
[ 8/May/2024 13:14:17] IN_CREATE /u01/dsmr/db/db.sqlite3-journal
[ 8/May/2024 13:14:17] IN_CLOSE_WRITE /u01/dsmr/db/db.sqlite3-journal
[ 8/May/2024 13:14:17] IN_DELETE /u01/dsmr/db/db.sqlite3-journal
[ 8/May/2024 13:14:17] * /u01/dsmr/db/db.sqlite3-journal is deleted
```

Doe ik iets verkeerd? Wat kan ik controleren zodat ik:

1. Geen "OperationalError" meer krijg,
2. als het mogelijk is, minder IO/s krijg[1]

Heeft iemand tips?

Alvast bedankt!

[1] 1 van mijn SSDs geeft SMART errors omdat het inmiddels boven het aantal TBW komt en dat wil ik voorkomen op de volgende SSD. 

### DSMR-reader version

v5.11.0

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-05-09:
Bedankt voor je melding. Weet je zeker dat je PostgreSQL gebruikt? Dit gezien de melding met `dsmr/db/db.sqlite3-journal`.

Wat betreft optimalisatie. Ik denk dat het inherent is aan een database gebruiken. Wellicht is er voor PostgreSQL wat in te stellen dattie minder vaak naar de schijf schrijft, maar dat valt echt buiten dit project. 

Want dan kom je echt bij de niet-standaard instellingen, zoals https://dba.stackexchange.com/a/287126, en daar heb ik me zelf ook nooit mee bezig gehouden.

mjholtkamp, 2024-05-09:
Hoi @dennissiemensma, ook jij bedankt voor het kijken naar mijn issue.

Volgens mij heb ik geen PostgreSQL genoemd, alleen sqlite 😅. Ik weet eigenlijk niet of sqlite officieel ondersteund wordt, ik heb dat gekozen omdat het weer een service scheelt (en dat het tot voor kort eigenlijk supergoed werkte).

dennissiemensma, 2024-05-09:
Dat verklaart een hoop. Sqlite wordt niet ondersteund, wel door het onderliggende code framework, maar niet functioneel. PostgreSQL is de enige officiële database voor DSMR-reader, al het andere werkt wel (of niet), maar wordt helaas niet ondersteund.

mjholtkamp, 2024-05-09:
Ah, als het niet ondersteund wordt, dan weet ik nu dus wat ik verkeerd heb gedaan.

Voor jouw informatie; ik heb SQLite meerdere jaren gebruikt, maar waarschijnlijk loop ik nu dus tegen een performanceprobleem aan.

In ieder geval ga ik het dan toch maar omzetten naar PostgreSQL, dank!

closed 2024-05-09 by mjholtkamp

mjholtkamp, 2024-05-09:
Omzetten naar PostgreSQL loste het probleem met de locked database op, het is nu weer responsive!