dsmrreader/dsmr-reader #1999: MQTT: Rejecting message for topic due to maximum queue size (5000)

edgewall43, 2024-06-03:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Gisteravond stopte DSMR reader plotseling met het versturen wan de MQTT berichten naar de Mosquitto broker in Home Assistant. 
Melding in de web portal: Er zijn te veel uitgaande MQTT-berichten die wachten op verzending 

Logging van Mosquitto geeft aan wel een verbinding tot stand te krijgen maar MQTT wordt niet verzonden.

### DSMR-reader version

5.11

### DSMR-reader platform

Other (e.g. Home Assistant Plugin)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
MQTT: Rejecting message for topic due to maximum queue size (5000)
```

dennissiemensma, 2024-06-03:
Zie
- https://github.com/dsmrreader/dsmr-reader/issues/1845#issuecomment-1542463972
- https://github.com/dsmrreader/dsmr-reader/issues/1643#issuecomment-1152522679

Voor de exacte oorzaak zul je echt de logs in moeten. Zoals bij de bovenste link. Dan zie je ook wat/wanneer iets gebeurt.

edgewall43, 2024-06-04:
Bedankt, na het uit- en weer inschakelen van alle MQTT databronnen functioneert het weer naar behoren en lijken alle topics in de queue ook verwerkt te zijn. (deze actie uitgevoerd rond 23.30 uur en nieuwe waarden vanaf 21.00 uur in HomeAssistant beschikbaar).

closed 2024-06-04 by edgewall43