dsmrreader/dsmr-reader #2000: READ! MQTT Broker; no incoming data from DSMR Reader, fix!

neforce, 2024-06-12:
### Language / Voertaal

🇬🇧 English

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Hi there,

Suddenly my mqtt doesn't receive any messages anymore. After that, i also migrated the whole thing to another NUC. But the mqtt broker still doesn't receive anything from dsmr reader (tho, it receives from other software).

I'm seeing this in the logs of the mqtt broker:

2024-06-12 13:27:56: New client connected from 192.168.XX.XX:51573 as DSMR-reader (p2, c1, k60, u'USERNAME').
2024-06-12 13:29:26: Client DSMR-reader has exceeded timeout, disconnecting.

But in the logs of the dsmr_backend.log i see:

```
2024-06-12 14:26:24,711 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: integer out of range
2024-06-12 14:27:25,744 ERROR    apps         _on_electricity_consumption_created_signal   94 | publish_day_consumption() failed: integer out of range

```

After that, i'd checked the database with HeidiSQL. And by change i found the issue.
I entered in table dsmr_mqtt_message an entry. But after saving it displayed the same error: integer out of range.

So i fixed the auto increment:

SELECT setval(pg_get_serial_sequence('dsmr_mqtt_message', 'id'), 1, false);

And that did the trick! I'm pretty sure the auto increment value exeeds the integer limit after 3-4 years of publishing.

Regards!

Leon

### DSMR-reader version

v5.11

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
DSMR-READER
    App | Python | Database                                                     v5.11 | v3.10.12 | postgresql
    BE sleep | DL sleep | Retention | Override                                     1.0s | 2.0s | 744h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                            2728468
    Consumption records electricity | gas (est.)                                               113691 | 77613
```

### Optional: Smart meter telegram

```shell-script
/Ene5\XS210 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(240612134012S)
0-0:96.1.1(4530303437303030303530393930303138)
1-0:1.8.1(009715.994*kWh)
1-0:1.8.2(008350.965*kWh)
1-0:2.8.1(004280.166*kWh)
1-0:2.8.2(009211.823*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.033*kW)
0-0:96.7.21(00226)
0-0:96.7.9(00004)
1-0:99.97.0(3)(0-0:96.7.19)(230311160108W)(0000004691*s)(210818200550S)(0000001451*s)(181201231405W)(0000000543*s)
1-0:32.32.0(00004)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(235.0*V)
1-0:31.7.0(005*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(01.033*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303732303033393237333331373139)
0-1:24.2.1(240612134000S)(03329.571*m3)
!8720
```

dennissiemensma, 2024-06-12:
Great, thanks for sharing!

It's a known flaw and the solution is moving from integers to UUID's, but it's quite some work to implement, so your workaround should suffice, for now!

closed 2024-06-12 by dennissiemensma

jhoekstra, 2024-08-10:
@neforce This workaround helped me fix the same issue on an instance which has been running since early 2018. Thanks for the hint!

wimb0, 2025-04-20:
Thanks, I've had the same issue here with my setup running since march 2018 since early this morning.
This workaround fixed it.