dsmrreader/dsmr-reader #2001: Not able to vonfigure dmsr-reader

todool, 2024-06-14:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

I have the dsmr reader running on a Raspberry PI4 (Home assistant) via docker. If I try to open the configuration page I get the message : 
"Forbidden (403)
CSRF verification failed. Request aborted.

You are seeing this message because this HTTPS site requires a “Referer header” to be sent by your Web browser, but none was sent. This header is required for security reasons, to ensure that your browser is not being hijacked by third parties.

If you have configured your browser to disable “Referer” headers, please re-enable them, at least for this site, or for HTTPS connections, or for “same-origin” requests.

If you are using the <meta name="referrer" content="no-referrer"> tag or including the “Referrer-Policy: no-referrer” header, please remove them. The CSRF protection requires the “Referer” header to do strict referer checking. If you’re concerned about privacy, use alternatives like <a rel="noreferrer" …> for links to third-party sites."

The access to my raspberry is via DuckDNS (https)

Whatever I do, other browser, cleaning cookies and so on nothing helps.

### DSMR-reader version

1.11.3

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-06-14:
Bedankt voor je vraag. Allereerst, draai je daadwerkelijk nog DSMR-reader v1? 

Wat is de dump van: https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/debug-info-dump.html

dennissiemensma, 2024-06-14:
Voor de rest heb ik overigens geen idee wat het precies veroorzaakt. Mocht je nog wel een hele oude DSMR-reader draaien, dan is updaten denk ik stap 1.

Ik dacht ook nog even aan dit issue, maar dat lijkt een andere fout: https://github.com/dsmrreader/dsmr-reader/issues/344#issuecomment-315523564

ov-dl, 2024-06-21:
Hi Dennis,

Ik heb hetzelfde probleem met dsmr reader in docker (xirixiz/dsmr-reader-docker:latest).
Debug output:
DSMR-READER
    App | Python | Database                                                      v5.11 | v3.11.9 | postgresql
    BE sleep | DL sleep | Retention | Override                                    1.0s | 0.5s | 2232h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                            2794775
    Consumption records electricity | gas (est.)                                              213903 | 104914

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                1956 MB

Via homepage -> configuration -> inlog form, creds invullen ->
Forbidden (403)
CSRF verification failed. Request aborted.
More information is available with DEBUG=True.

Melding is hetzelfde ongeacht of creds goed of fout zijn.

$ docker ps
CONTAINER ID   IMAGE                                    COMMAND                  CREATED          STATUS                    PORTS                                                                                                                                                           NAMES
6562d6719152   xirixiz/dsmr-reader-docker:latest        "/init"                  10 minutes ago   Up 10 minutes (healthy)   0.0.0.0:7777->80/tcp, :::7777->80/tcp, 0.0.0.0:7779->443/tcp, :::7779->443/tcp

Ik heb de volgende vars al geprobeerd door deze toe te voegen aan de docker-compose.yaml ("environment:" sectie onder 'dsmr'), maar zonder succes:
      - CSRF_TRUSTED_ORIGINS="https://192.168.200.251:7000,https://localhost:7000"
      - DJANGO_USE_X_FORWARDED_HOST=true
      - DEBUG=True

Als je meer info nodig hebt, let me know!

Gr.
Onno.

Dickey01, 2024-06-27:
Hallo Onno, 

Heb hetzelfde probleem en config. Heb je het probleem al kunnen achterhalen?

Gr.
Dick

dennissiemensma, 2024-06-30:
@ov-dl 

> _! EDIT: Hmm, kom er net achter dat ik zelf er een nginx proxy voor had gezet voor SSL. De foutmelding komt dus wanneer ik hem via nginx open. Als ik rechtstreeks ga (zonder nginx/ssl) dan krijg ik de foutmelding niet. Ik zal de komende dagen eens verder spitten._

Gezien de foutmelding:

```
You are seeing this message because this HTTPS site requires a “Referer header” to be sent by your Web browser, but none was sent. This header is required for security reasons, to ensure that your browser is not being hijacked by third parties.

If you have configured your browser to disable “Referer” headers, please re-enable them, at least for this site, or for HTTPS connections, or for “same-origin” requests.
```

Kun je kijken of je browser wel die header stuurt. Of dat er iets in je browser voorkomt dat het gestuurd wordt. Of dat Nginx die `Referer` header wel doorstuurt naar DSMR-reader.

the1rig, 2024-07-01:
I'm running into the same issue, but then on the docker version for Homeasssistant. The same issue is open on the related git page: https://github.com/sanderdw/hassio-addons/issues/95

Dickey01, 2024-07-02:
Hallo Dennis,De referer wordt wel meegestuurd.Ik heb het vermoeden dat het is ontstaan na de laatste update.Groetjes DickVerstuurd vanaf mijn iPhone

dennissiemensma, 2024-07-02:
Het had ook nog dit kunnen zijn, via `CSRF_TRUSTED_ORIGINS`, zoals @ov-dl al aangeeft:
- https://stackoverflow.com/a/70509982

Echter [gebruikt DSMR-reader v5.11 nog Django 3.2](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmrreader/provisioning/requirements/base.txt#L12). 

Daar zit die instelling al wel in, [in oudere vorm](https://docs.djangoproject.com/en/3.2/ref/settings/#csrf-trusted-origins), maar DSMR-reader ontsluit die (nog) niet. Het meegeven van die envvar doet niets, alleen het hardcoded opnemen in de configs.

----

Dus dat gaat overigens wel iets worden voor DSMR-reader v6.x, die een flinke stap maakt qua minimale versies en wel Django 4.x gaat gebruiken. Dus daar zal ik zorgen voor een envvar ter voorbereiding. Zodat `CSRF_TRUSTED_ORIGINS` in te stellen is.

En wellicht nog iets voor de laatste DSMR-reader v5.12.

remb0, 2024-07-06:
Dat is goed nieuws dennis!  kan ik mijn contracten weer aanpassen ;)

dennissiemensma, 2024-07-08:
Voor de duidelijkheid: Ik weet niet zeker of het issue daar in zit. 

Als je de bestanden van je DSMR-reader installatie kunt aanpassen, kun je proberen om `CSRF_TRUSTED_ORIGINS` in te stellen in een `.env`-bestand in de bovenste map van je installatie (bij native installaties). Waar ook deze staat: [.env.template](https://github.com/dsmrreader/dsmr-reader/blob/v5/.env.template). En daarma `./reload.sh` uitvoeren als `dsmr` user.

Voor Docker-installaties is het wat lastiger, omdat die envvar niet gemapped wordt. Maar wellicht is daar een `.env` bestand ook wel te maken (bestaat wellicht al), al zul je dan echt degene binnen de container moeten hebben en niet degene daarbuiten, die evt door Docker zelf gebruikt wordt.

Zie mijn vorige comment voor wat voorbeelden.

TerrorSource, 2025-01-06:
@dennissiemensma Is hier al een oplossing voor gevonden?
Draai de DSMR docker in HomeAssistant. Via HA extern en intern, beide via https, kom ik niet bij de configuratie.

``
Forbidden (403)
CSRF verification failed. Request aborted.

You are seeing this message because this HTTPS site requires a “Referer header” to be sent by your Web browser, but none was sent. This header is required for security reasons, to ensure that your browser is not being hijacked by third parties.

If you have configured your browser to disable “Referer” headers, please re-enable them, at least for this site, or for HTTPS connections, or for “same-origin” requests.

If you are using the <meta name="referrer" content="no-referrer"> tag or including the “Referrer-Policy: no-referrer” header, please remove them. The CSRF protection requires the “Referer” header to do strict referer checking. If you’re concerned about privacy, use alternatives like <a rel="noreferrer" …> for links to third-party sites.
``

dennissiemensma, 2025-01-06:
@TerrorSource ik kan het helaas niet makkelijk reproduceren. Echter zal de oplossing zitten in het instellen van 1 of meerdere waardes in de config van Django (= onderliggende framework).

Je kunt deze voor nu proberen: https://github.com/dsmrreader/dsmr-reader/issues/2001#issuecomment-2214999909

Een structurele fix om via envvars te spelen met die instellingen wacht op DSMR-reader 6.x

TerrorSource, 2025-01-07:
> @TerrorSource ik kan het helaas niet makkelijk reproduceren. Echter zal de oplossing zitten in het instellen van 1 of meerdere waardes in de config van Django (= onderliggende framework).
> 
> Je kunt deze voor nu proberen: [#2001 (comment)](https://github.com/dsmrreader/dsmr-reader/issues/2001#issuecomment-2214999909)
> 
> Een structurele fix om via envvars te spelen met die instellingen wacht op DSMR-reader 6.x

Ik heb een soort-van workaround nu:

Webserver: "Standaard" (dus niet "Ingress")
Networkport: 7777

en dan op IPvanHA:7777 kan je er bij en kan er wel ingelogd worden in Admin gedeelte.
De viewer binnen HA werkt dan alleen niet meer goed.

dennissiemensma, 2025-01-07:
Een hoop settings zijn toegevoegd voor DSMR-reader v6
https://github.com/dsmrreader/dsmr-reader/commit/3757f56441770f4e48fca4982ec969939433fa65#diff-c7e9be43431a27c0ea637364c35c601d25236877a8ea126f379b544182e60bd6

Ondanks dat ik niet zeker weet of dit het issue fixt, ga ik er wel vanuit, gezien het een kwestie van whitelisten lijkt in Django voor CSFR.

closed 2025-01-07 by dennissiemensma

dennissiemensma, 2026-02-01:
@DarkShadowNight welke DSMR-reader versie gebruik je? En welke [DJANGO_CSRF_* instellingen](https://dsmr-reader.readthedocs.io/en/development/reference/environment-variables/#django_csrf_cookie_age)? 

Er is hier geen globale fix voor, het is meestal een per-setup of zelfs per-gebruiker set aan configuratie dat nodig is.

Zie ook deze thread: https://github.com/dsmrreader/dsmr-reader/discussions/2098#discussioncomment-15454895

Fonta, 2026-02-16:
Had hier zelfde issue, na upgraden naar v6 kreeg ik na inloggen op het admin panel een 403 forbidden.
Mijn vermoeden is dat dit te maken heeft met de port mapping in docker.
Ik draai dsmr op mijn nas en heb de 7777:80 mapping.
Na toevoegen van de DJANGO_CSRF_TRUSTED_ORIGINS environment variabele kan ik er weer in.
Enkel url toevoegen was niet genoeg. Moest inclusief poort zijn.

[image]