dsmrreader/dsmr-reader #2005: P1 telegram sturen naar DSMR Reader met mqtt

mchangsp, 2024-06-30:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Voeg als datalogger / input methode een nieuwe methode toe: DSMR Reader luistert naar een mqtt topic waarop een P1 telegram word gepubliceerd.
Deze methode van input leveren is eenvoudiger dan het schrijven van plugins.

### DSMR-reader version

V5.11

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-06-30:
Bedankt voor je verzoek. Ik heb hier eerder naar gekeken en het heeft helaas teveel haken en ogen:

- https://github.com/dsmrreader/dsmr-reader/issues/1361#issuecomment-818920133

Als ik het project opnieuw zou beginnen dan zou ik het heroverwegen. DSMR-reader is oorspronkelijk nooit ontworpen als dataontvanger, puur als databron. Vandaar.

mchangsp, 2024-07-02:
In plaats van MQTT gebruiken heb ik het volgende gedaan. Misschien kan jet het ergens in de DSNR Reader FAQ neerzetten? https://community.home-assistant.io/t/how-to-get-homewizard-p1-meter-telegrams-into-dsmr-reader-using-automations/745345

dennissiemensma, 2024-07-02:
@mchangsp thanks! Ik zal het opnemen in de docs voor DSMR-reader v6.x

mchangsp, 2024-07-02:
Dank je! Ik ben benieuwd wat er in v6.x komt.

dennissiemensma, 2024-07-02:
De grote versie-bumps zijn bijna altijd de momenten waarop er veel onderliggende technische wijzigingen zijn. Inhoudelijk zal DSMR-reader in 6.0 niet meteen anders zijn. Wellicht wel wat sneller als gevolg van upgrades.

- https://github.com/dsmrreader/dsmr-reader/milestone/146?closed=1
- https://github.com/dsmrreader/dsmr-reader/milestone/146

marcovn, 2024-12-14:
Ik heb nog een alternatief voor mensen die, net zoals ik, de homewizard p1 meter gebruiken. Je kunt met de onderstaande one-liner (shell script) een netwerk socket maken die DSMRreader gewoon standaard kan uitlezen. 

Het lees een telegram van de p1meter (via local api) elke 5 seconden. De output gaat via een pipe naar een network socket, poort 3334 in mijn geval. De host/port combinatie vul je in de datalogger configuratie.

Je kunt dit script draaien in een screen sessie of via een systemd service als service laten draaien.

```
#!/bin/bash

while true; do
    curl http://<<hostnaam of ip-adres van je p1 meter>>/api/v1/telegram -s
    sleep 5
done | socat - TCP-LISTEN:3334,reuseaddr,fork

```

dennissiemensma, 2025-01-07:
Toekomstige docs bijgewerkt:

https://dsmr-reader.readthedocs.io/en/work-in-progress-v6/how-to/third-party/third-party-guides/#third-party-guides

Onderdeel van #1686

closed 2025-01-07 by dennissiemensma