dsmrreader/dsmr-reader #2006: welke body meegeven in opsturen P1 telegram met V1 api http://[ip]/api/v1/datalogger/dsmrreading

mchangsp, 2024-07-01:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Ik krijg HTTP error 400 als ik de body hieronder gebruik in een POST http://[ip]/api/v1/datalogger/dsmrreading
Wat moet ik precies meesturen? (In mijn DSMR Reader instance kan ik de V1 API documentatie helaas niet vinden)
* ik haal een P1 telegram op uit de Homewizard P1 meter met GET http://[ip Homewizard meter]/api/v1/telegram
* Ik zet de tekst "telegram: " en krijg zo een nieuwe body
* ik stuur de body op met een Authorization header naar http://[ip DSMR Reader]/api/v1/datalogger/dsmrreading

Zie hieronder een voorbeeld
De "0-0:96.1.1" regel heb ik weggehaald.
QUOTE
telegram: /KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(240701132703S)
1-0:1.8.1(015175.791*kWh)
1-0:1.8.2(013644.303*kWh)
1-0:2.8.1(002422.369*kWh)
1-0:2.8.2(004913.148*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.379*kW)
0-0:96.7.21(00001)
0-0:96.7.9(00001)
1-0:99.97.0(2)(0-0:96.7.19)(200113201326W)(0000014909*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(006*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(01.379*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353631323533313738333135)
0-1:24.2.1(240701130000S)(09330.695*m3)
!29E6
ENDQUOTE

### DSMR-reader version

v5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-07-01:
Je kunt de API-docs vinden onder `/support` in je installatie. Of direct onder `/docs/api/redoc`. Er staat ook nog een Postman collectie in.

[image]

----

Dat zeggende hebben, is dat de v2 API docs en je zoekt in jouw geval v1. Ik dacht dat ik daar docs van had, maar ik kan ze niet zo snel vinden. 

Overigens zit daar maar 1 endpoint in en die kun je op deze manier aanroepen:
- https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_api/tests/v1/test_api.py#L26-L52

Voor de leesbaarheid staat de body daar over meerdere regels, maar het is 1 lange string. Inclusief de `/r/n` tekens.
Het zou gewoon direct het telegram moeten zijn uit je meter, inclusief de checksum aan het einde.

dennissiemensma, 2024-07-01:
Ik heb je voorbeeld bijgewerkt en ook het nummer van je gasmeter weggehaald.
Het zou ongeveer zo moeten werken, mits de checksum goed is uiteraard.

mchangsp, 2024-07-01:
Dank je wel Dennis (iik voor het weghalen van mijn gasmeter nummer)!
Ik ga verder experimenteren.

dennissiemensma, 2024-07-01:
Als ik het voorbeeld uit de test pak, dan moet die zo verstuurd worden:

[image]

----

[image]

----

Ik krijg welliswaar deze fout, maar dat ligt aan dat telegram zelf:
```
2024-07-01 19:48:37,589 DEBUG    datalogger   telegram_to_reading              81 | Received telegram:
2024-07-01T17:48:37.590033302Z /KFM5KAIFA-METER\r\n\r\n1-3:0.2.8(42)\r\n0-0:1.0.0(160303164347W)\r\n0-0:96.1.1(12345678901234567890123456789012)\r\n1-0:1.8.1(001073.079*kWh)\r\n1-0:1.8.2(001263.199*kWh)\r\n1-0:2.8.1(000000.000*kWh)\r\n1-0:2.8.2(000000.000*kWh)\r\n0-0:96.14.0(0002)\r\n1-0:1.7.0(00.143*kW)\r\n1-0:2.7.0(00.000*kW)\r\n0-0:96.7.21(00006)\r\n0-0:96.7.9(00003)\r\n1-0:99.97.0(1)(0-0:96.7.19)(000101000001W)(2147483647*s)\r\n1-0:32.32.0(00000)\r\n1-0:32.36.0(00000)\r\n0-0:96.13.1()\r\n0-0:96.13.0()\r\n1-0:31.7.0(000*A)\r\n1-0:21.7.0(00.143*kW)\r\n1-0:22.7.0(00.000*kW)\r\n!4BC6\n
2024-07-01T17:48:37.591504254Z 2024-07-01 19:48:37,591 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '54250' (D3EA) does not match the telegram checksum '19398' (4BC6)
2024-07-01T17:48:37.591669647Z Internal Server Error: /api/v1/datalogger/dsmrreading
2024-07-01T17:48:37.591856086Z [01/Jul/2024 19:48:37] "POST //api/v1/datalogger/dsmrreading HTTP/1.1" 500 24
```

dennissiemensma, 2024-07-01:
Met nog de aanvulling dat niet alle headers nodig zijn. Postman vult die zelf aan. De `Content-Type` en `Authorization` zijn wel nodig.

De tekst `telegram: ` is dus het body-veld `telegram`, geen prefix aan de body.

closed 2024-07-01 by dennissiemensma