dsmrreader/dsmr-reader #2011: geen data meer (??)

newtech2015, 2024-07-03:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Hoi,

ik zie opeens, dat ik geen data meer krijg in dsmr, vanuit mijn meter (enexis), al 3dg en 10 uur.... 
weet iemand mss, of enexis, dit heeft uitgeschakeld?  vind het wel erg toevallig dat dit zo ongeveer begonnen is op 1 juli.

gr ron

### DSMR-reader version

5.8.0

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-07-04:
Bedankt voor je melding. Je kunt het beste wat meer informatie aanleveren, zoals deze dump: https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/debug-info-dump.html

En wat geven de [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) verder aan?

newtech2015, 2024-07-05:
dump:
DSMR-READER
    App / Python / Database                                                        v5.8 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                   1.0s / 10.0s / 2016h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             757279
    Consumption records electricity / gas (est.)                                              201873 / 107554

UNRESOLVED ISSUES
    No recent readings received                                                          4 days, 22 hours ago
    Day statistics are lagging behind                                                     5 days, 9 hours ago

sudo supervisorctl status
dsmr_backend                     RUNNING   pid 572, uptime 1 day, 11:37:26
dsmr_datalogger                  RUNNING   pid 23211, uptime 0:00:01
dsmr_webinterface                RUNNING   pid 574, uptime 1 day, 11:37:25

dsmr_backend
-bash: dsmr_backend: command not found
 ackend: command not found
dsmr_datalogger
-bash: dsmr_datalogger: command not found
 dsmr_webinterface
-bash: dsmr_webinterface: command not found

newtech2015, 2024-07-05:
dan, lijkt mij de datalogger, vreemd doen, log:
urrent logging level set to "ERROR". More information can be found here: https>
2024-07-05 08:14:07,915 ERROR    mixins       run_once                        1>
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/serial/serialp>
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_clie>
    serial_handle = serial.serial_for_url(
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/serial/__init_>
    instance.open()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/serial/serialp>
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._>
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: >

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

wat betekend dit precies??

closed 2024-07-05 by newtech2015

newtech2015, 2024-07-05:
dennis, 

heel appart, vanmiddag de serieel naar p1 adpter eens op mijn pc aangesloten, hij zag een device...
daarna terug in pi en meter gedaan, en... het werkt weer!

zie nu ook een extra device in pi, tov vanochtend:
Bus 001 Device 017: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC

dennissiemensma, 2024-07-11:
Oke fijn om te horen!

De initiele melding gaf inderdaad aan dat die de datalogger/USB-poort niet kon vinden. Maar ik had zelf ook geen workaround voor je, behalve herstarten of de device er uit en in pluggen.