dsmrreader/dsmr-reader #2013: Too many outgoing InfluxDB measurements queued for transit

theautomation, 2024-07-10:
### Language / Voertaal

🇬🇧 English

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Hi, I had setup a InfluxDB instance to transfer measurements to. That worked fine, but now I deleted my InfluxDB instance because I no longer want to use it. First I deleted the InfluxDB instance while the InfluxDB integration in DSMR-reader was still enabled and now it complains about an issue. I am aware that it would have been better if I had done this the other way around.

> Too many outgoing InfluxDB measurements queued for transit   (2 days, 14 hours ago)

How can I solve this issue and get rid of the **Too many outgoing InfluxDB measurements queued for transit   (2 days, 14 hours ago)** message?

It would be nice to stop sending data to influxdb, now and in the past, if the integration is disabled.

BTW I have DMSR-reader stable as a rock for more than 4 years (Kubernetes) Thanks for your project! 

### DSMR-reader version

v5.11.0

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
DSMR-READER
    App | Python | Database                                                      v5.11 | v3.11.9 | postgresql
    BE sleep | DL sleep | Retention | Override                                     1.0s | 1.0s | 744h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                            1360683
    Consumption records electricity | gas (est.)                                               103991 | 67644

UNRESOLVED ISSUES
    Too many outgoing InfluxDB measurements queued for transit                           2 days, 14 hours ago

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 597 MB
```

wtreur, 2024-07-10:
I just had a similar issue.

Have you disabled the influx integration?

`Admin -> InfluxDB Integration -> Uncheck "Enabled" checkbox`

If already disabled, messages could still be pending in the queue. Remove them using the admin UI

`Admin -> Influxdb measurement -> Select all -> Select Delete from the action menu`

[image]

However, when removing large amount of messages, this issues occurs as well: #1643 In that case, you can manually remove the messages from the database:

```
sudo su - postgres
psql dsmrreader

delete from dsmr_influxdb_influxdbmeasurement;
```

theautomation, 2024-07-10:
Thank u very much! I had a lot of pages so I went for the manual deletion in the database and the issue message is now gone. 
```
root@dsmr-reader-postgresql-0:/# psql -U dsmrreader -d dsmrreader -h 127.0.0.1 -p 5432
Password for user dsmrreader: 
psql (14.2)
Type "help" for help.

dsmrreader=> delete from dsmr_influxdb_influxdbmeasurement;
DELETE 3400
dsmrreader=> 
```

closed 2024-07-10 by theautomation

dennissiemensma, 2024-07-11:
Great to hear, thanks for the  assistance as well! I'm rewriting/simplifying the docs, so this issue might be a good candidate for highlighting in the docs.