dsmrreader/dsmr-reader #2014: No readings after meter change

Martinvdm, 2024-07-11:
### Language / Voertaal

🇬🇧 English

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

After meter change no readings. Changed reader from DMSR2/3 to 4/5. but no readings.
Checked the support site, waited for 1 day, still no readings. recreated container etc.

### DSMR-reader version

5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
2024-07-11 09:32:37,007 ERROR    mixins       run_once                        107 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/app/dsmr_backend/mixins.py", line 96, in run_once
    self.run(data=self.data, **options)
  File "/app/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 51, in read_telegram
    raise RuntimeError(
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0
```

Martinvdm, 2024-07-11:
Fixed with new connector

closed 2024-07-11 by Martinvdm

dennissiemensma, 2024-07-11:
Thanks for the swift update, the error simply indicated that there was no data available, but it's quite a hassle to find the root cause of it. Good find and fix by yourself!