dsmrreader/dsmr-reader #2015: DSMR-reader not sending MQTT and lagging web server

p4r4ll4x, 2024-07-13:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

I'm running DSMR-reader in docker.
My server was offline for several hours today.

After i booted it again, DSMR-reader starts fine.
Messages from my smart reader are correctly received on the serial USB, checked on the host.
Log file show nothing wrong

### DSMR-reader version

5.11.0

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service init-docker-entrypoint: starting

[ INFO ] DSMR Reader Reader release: 5.11.0

[ INFO ] DSMR Reader Reader Docker release: 2024.04.03
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
    ───────────────────────────────────────────────────
     ___  ___ __  __ ___   ___             _
    |   \/ __|  \/  | _ \ | _ \___ __ _ __| |___ _ _
    | |) \__ \ |\/| |   / |   / -_) _` / _` / -_) '_|
    |___/|___/_|  |_|_|_\ |_|_\___\__,_\__,_\___|_|
                 __           _
                |   \ ___  __| |_____ _ _
                | |) / _ \/ _| / / -_) '_|
                |___/\___/\__|_\_\___|_|
    DSMR-protocol reader, telegram data storage and
            energy consumption visualizer.
    ───────────────────────────────────────────────────
    User UID: 803
    User GID: 803
    ───────────────────────────────────────────────────
    

[ INFO ] Ensure permissions...

[ INFO ] Setting architecture requirements...

[ INFO ] X64 Architecture

[ INFO ] Verifying if the DSMR Reader web credential variables have been set...

[ INFO ] Verifying database connectivity to host: 10.0.0.5 with port: 5433...

[ INFO ] Database connectivity successfully verified!

[ INFO ] Running post configuration...
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_dropbox, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.
  Your models in app(s): 'dsmr_mqtt' have changes that are not yet reflected in a migration, and so won't be applied.
  Run 'manage.py makemigrations' to make new migrations, and then re-run 'manage.py migrate' to apply them.
0 static files copied to '/var/www/dsmrreader/static', 559 unmodified.
Updating password of superuser "admin"
Deactivating any other existing superusers

[ INFO ] Checking for CLIENT CERTIFICATE AUTHENTICATION configuration...

[ INFO ] ENABLE_CLIENTCERT_AUTH is disabled, nothing to see here. Continuing...

[ INFO ] Checking for NGINX SSL configuration...

[ INFO ] ENABLE_NGINX_SSL is disabled, nothing to see here. Continuing...

[ INFO ] Checking for HTTP AUTHENTICATION configuration...

[ INFO ] ENABLE_HTTP_AUTH is disabled, nothing to see here. Continuing...

[ INFO ] Configuring DSMR Reader to run all processes in a single container with a serial or ipv4 datalogger...

[ INFO ] Using a serial connection for the DSMR Reader remote datalogger...

[ INFO ] Fixing /dev/ttyUSB0 security...

[ INFO ] Adding serial input method configuration settings...
s6-rc: info: service init-docker-entrypoint successfully started
s6-rc: info: service svc-dsmr-webinterface: starting
s6-rc: info: service svc-dsmr-remote-datalogger: starting
s6-rc: info: service svc-dsmr-datalogger: starting
s6-rc: info: service svc-dsmr-backend: starting
s6-rc: info: service svc-dsmr-webinterface successfully started
s6-rc: info: service svc-nginx: starting
Starting DSMR Reader - webinterface...
Starting DSMR Reader - backend...
s6-rc: info: service svc-dsmr-webinterface-log: starting
s6-rc: info: service svc-dsmr-remote-datalogger successfully started
s6-rc: info: service svc-dsmr-datalogger successfully started
s6-rc: info: service svc-dsmr-datalogger-log: starting
s6-rc: info: service svc-dsmr-backend successfully started
s6-rc: info: service svc-dsmr-backend-log: starting
s6-rc: info: service svc-nginx successfully started
Starting DSMR Reader - datalogger...
s6-rc: info: service svc-dsmr-webinterface-log successfully started
s6-rc: info: service svc-dsmr-backend-log successfully started
s6-rc: info: service svc-dsmr-datalogger-log successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[2024-07-13 18:37:21 +0000] [106] [INFO] Starting gunicorn 20.1.0
[2024-07-13 18:37:21 +0000] [106] [INFO] Listening at: unix:/tmp/gunicorn--dsmr_webinterface.socket (106)
[2024-07-13 18:37:21 +0000] [106] [INFO] Using worker: sync
[2024-07-13 18:37:21 +0000] [145] [INFO] Booting worker with pid: 145

Testing webservice socket connectivity: 1 second(s) of 10 seconds...Starting DSMR Reader - nginx...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2024-07-13 20:37:27,709 ERROR    services     get_dropbox_client               62 |  - Dropbox error: HTTPSConnectionPool(host='api.dropboxapi.com', port=443): Max retries exceeded with url: /oauth2/token (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f402e95b1d0>: Failed to establish a new connection: [Errno -3] Try again'))
2024-07-13 20:37:27,711 ERROR    schedule     execute_scheduled_processes      39 | (ConnectionError) dsmr_dropbox.services.run errored: HTTPSConnectionPool(host='api.dropboxapi.com', port=443): Max retries exceeded with url: /oauth2/token (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f402e95b1d0>: Failed to establish a new connection: [Errno -3] Try again')) 
127.0.0.1 - - [13/Jul/2024:18:37:32 +0000] "GET /about HTTP/1.1" 200 14040 "-" "curl/8.5.0" "-"
127.0.0.1 - - [13/Jul/2024:18:37:48 +0000] "GET /about HTTP/1.1" 200 13475 "-" "curl/8.5.0" "-"
127.0.0.1 - - [13/Jul/2024:18:38:33 +0000] "GET /about HTTP/1.1" 200 13475 "-" "curl/8.5.0" "-"
127.0.0.1 - - [13/Jul/2024:18:38:48 +0000] "GET /about HTTP/1.1" 200 13475 "-" "curl/8.5.0" "-"
10.0.0.36 - - [13/Jul/2024:18:39:22 +0000] "GET / HTTP/1.1" 499 0 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" "-"
10.0.0.16 - - [13/Jul/2024:18:39:33 +0000] "GET / HTTP/1.1" 499 0 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" "10.0.0.36"
```

dennissiemensma, 2024-07-15:
Please provide a debug dump to see if there are any issues: https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/debug-info-dump.html

The docker logs do not show any issues. Most common reason for lagging is too much data stored, which the above will show (if you did not suppress it).

p4r4ll4x, 2024-07-15:
> Please provide a debug dump to see if there are any issues: https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/debug-info-dump.html
> 
> The docker logs do not show any issues. Most common reason for lagging is too much data stored, which the above will show (if you did not suppress it).

I restarted the morning after and everything is fine again.
Could it be that he was processing the data from the missed hours, if that is even possible?

dennissiemensma, 2024-07-15:
Thanks, it's must be something else then. Those data numbers are not that big, far from it.
And if the data was lagging behind, it will also show on that same dump.

I guess you could try some more in-dept debugging if it happens again, but it's quite advanced: https://github.com/dsmrreader/dsmr-reader/issues/248#issuecomment-276474980

closed 2024-08-16 by dennissiemensma