dsmrreader/dsmr-reader #2024: correctie op 1 dag nar nieuwe meter

remb0, 2024-09-21:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

op maandag 5 auguster heb ik een nieuwe beter gekregen.   
mijn dsmrreading veranderden maar nu krijg ik geen kloppende statistieken meer 

[image]
[image]

Ik heb alleen geen enkel idee hoe en waar ik dit moet corrigeren om weer een kloppend maand/jaar totaal te kunnen zien.

### DSMR-reader version

v5.11

### DSMR-reader platform

Other (e.g. Home Assistant Plugin)

dennissiemensma, 2024-09-21:
Zie https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/smart-meter-replacement.html

remb0, 2024-09-21:
dank voor de link. Ik snap dat ik iets moet corrigeren. maar weet niet zeker wat.
de foutieve dag verwijderen?
[image]

of maak ik het dat erger?

dennissiemensma, 2024-09-21:
Je kunt de foute dagtotalen van de dag van de wissel met de hand op 0 zetten. Of ze zelf uitrekenen voor die dag. Alles na die dag zou verder al goed moeten staan. En vervolgens ook automatisch alle maand en jaar totalen.

closed 2024-10-02 by dennissiemensma