dsmrreader/dsmr-reader #2025: Tailscale + HTTPS + DSRM-reader: How to contribute to documentation?

eparon, 2024-09-27:
### Language / Voertaal

🇬🇧 English

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

This is not an issue, but more of a question:

At my homelab, I use Tailscale to define my private mesh-VPN. Tailscale can issue for you SSL certificates for your network ([tailnet](https://tailscale.com/kb/1217/tailnet-name)), for which you are provided with a name, e.g. `tail-scale.ts.net`.

These HTTPS certificates can be used (e.g. https://dsmr-reader.readthedocs.io/en/v5/how-to/installation/https.html) to enable a secure connection with DSMR-reader. However, renewing the Tailscale HTTPS certs is a manual action.

The web server [Caddy](https://caddyserver.com/) supports native integration with Tailscale, which allows it to automatically obtain AND renew the HTTPS certs, whenever needed.

I recently experimented on integrating Caddy with DSMR-reader and have documented the required steps. As this was not (fully) trivial, it could be beneficial to add these steps in the projects documentation.

What would be the best way to do this? (If, of course, there is interest for something like that)
Make a PR and update the `docs` by adding a new `howto` page? 
Add it in the repo's Discussions and tag it as a how-to guide?

### DSMR-reader version

latest

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2024-09-29:
Thanks for contributing!

At the moment DSMr-reader will transition into a more slim and Docker-only project, to reduce the support and extensive documentation a bit.

You could opt to host the docs yourself, like these do:
- https://github.com/xirixiz/dsmr-reader-docker
- https://github.com/sanderdw/hassio-addons

Then I would not mind to link to it in the docs eventually, similar to the ones above. It also allows you to update your own docs yourself.

However do know that if there are any support questions regarding such setup from other users, I will pass through the questions, as I cannot keep in sync with all the stuff used.

Also note that simply creating this issue here will also have Google index it eventually. It's up to you what you like most.

eparon, 2024-10-11:
Thanks for your explanation @dennissiemensma, I think it's better to document my steps here. I am also happy to answer any questions from others, should they appear.

## Tailscale
[Tailscale](https://tailscale.com/) is a mesh-VPN solution, which I use at home. It is based on WireGuard and offers a more 'civilized' way to manage your VPN. In addition to this, you can get a fully-qualified domain for your tailnet (your mesh-VPN network), where each device running Tailscale becomes addressable. E.g., assuming that my tailnet is `foo-bar.ts.net`, my raspberry-pi would be addressable at: `<rpi-hostname>.foo-bar.ts.net`. Tailscale allows you to expose different services (e.g. DSMR-reader) in your tailnet, without exposing them publicly on the internet. 

Bonus: Tailscale also provides you with SSL certificates for your tailnet's domain (and subdomains)!

## The setup
My raspberry-pi runs DSMR-reader (deployed directly on the host -- no container) and serves the web-page of DSMR-reader at port `:80`. Consequently, I used to access my DSMR-reader from my home network, using the IP of my raspberry-pi e.g. `http://192.168.1.5`. 

As I want to have trusted (i.e., not self-signed) SSL certificates at my home services, I considered:
* enrolling my raspberry-pi in my tailnet
* obtaining an SSL certificate from Tailscale for my tailnet
* configuring the SSL cert in the web server serving DSMR-reader's content.

**Caveat: when the SSL certs expire, I need to manually obtain new certs from Tailscale and reconfigure the web server** 

As this is laborious, I opted for automating this process using a web server called [Caddy](https://caddyserver.com/). Caddy integrates nicely with Tailscale and can automate the second and third bullets from the list above.

### Step 0/Prerequisites
You have Tailscale installed and configured.
You have your own tailnet.

### Step 1: Install Caddy
To do this on a raspberry-pi, you can run: 
```bash
sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https curl
curl -1sLf 'https://dl.cloudsmith.io/public/caddy/stable/gpg.key' | sudo gpg --dearmor -o /usr/share/keyrings/caddy-stable-archive-keyring.gpg
curl -1sLf 'https://dl.cloudsmith.io/public/caddy/stable/debian.deb.txt' | sudo tee /etc/apt/sources.list.d/caddy-stable.list
sudo apt update
sudo apt install caddy
```

### Step 2: Configure DMSR's NGINX to serve its content at a different port (not `80`)
To achieve this, you need to modify the file: `/etc/nginx/sites-available/dsmr-webinterface` and change line `listen      80;` to e.g. `listen      8080;`

### Step 3: Configure Tailscale to allow Caddy to issue SSL certificates 
Caddy runs with its own user. Tailscale must be configured to allow for Caddy's user. To achieve this, Tailscale's service configuration must be edited: `sudo vim /etc/default/tailscaled`
Add the following line at the end of the file: `TS_PERMIT_CERT_UID="caddy"`

Make sure to restart Tailscale: `sudo systemctl restart tailscaled.service`

### Step4: Configure Caddy 
Caddy's `Caddyfile`, will look like the following:
```Caddyfile
<tailscale-machine-name>.<tailnet-name-or-id>.ts.net {
	reverse_proxy <raspberry-pi-hostname>:8080
}
```
This approach will also work if you run DSMR-reader in a container -- make sure to `reverse_proxy` to the DNS name/port of DSMR-reader's container.

** Alternatively ** 

If you would prefer to not use NGINX as the 'main' web server for DSMR-reader (and you prefer to fully switch to Caddy), then your `Caddyfile` can look like:
```Caddyfile
<tailscale-machine-name>.<tailnet-name-or-id>.ts.net {

	handle_path /static/* {
		root * /var/www/dsmrreader/static
		file_server
	}

	reverse_proxy unix//tmp/gunicorn--dsmr_webinterface.socket
}
```

If you opt for this approach, do not forget to fully shut down NGINX on your system, or disable DSMR-reader's website.

Finally, reload Caddy: `sudo systemctl restart caddy.service`

closed 2024-10-14 by dennissiemensma