dsmrreader/dsmr-reader #2030: 5000 queued mqtt messages

bart1604, 2024-10-21:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Ik heb de laatste tijd veel issues gehad met de data die doorkomt naar mijn home assistant. Ik kreeg soms voor 2 minuten in home assistant geen updates binnen, en dan krijg ik weer 1 update en dan duurde het weer extreem lang
Ik ben wat gaan zoeken en kwam tegen dat een van de eerste dingen die kan checken is "Uitgaande MQTT-berichten"

Hier zag ik er heel de tijd weer 10-tallen bijkomen en weer afgaan maar zat ergens te schommelen tussen de 3300 en 3700 berichten. Ik dacht, dan zet ik alles uit. Dus
- alle databronnen uitvinkt -> geen verschil
- MQTT service binnen DSMR uitgeschakeld -> meldingen liepen op tot 5000.
- MQTT weer ingeschakeld -> meldingen lopen niet terug en krijg ook niks binnen in home assistant
- databronnen weer ingeschakeld -> krijg nog steeds niks nieuws meer binnen in home assistant

Ik weet niet zo goed wat ik nog kan doen?

### DSMR-reader version

v5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
DSMR-READER
    App | Python | Database                                                      v5.11 | v3.11.9 | postgresql
    BE sleep | DL sleep | Retention | Override                                     1.0s | 0.5s | 168h | False
    Latest telegram version read | Parser settings                                                 "42" | "4"

DATA
    Telegrams total (est.)                                                                              96981
    Consumption records electricity | gas (est.)                                                    96981 | -

UNRESOLVED ISSUES
    Er zijn te veel uitgaande MQTT-berichten die wachten op verzending                                      nu
```

### Optional: Smart meter telegram

```shell-script
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(241021143522S)
0-0:96.1.1(4530303236303030303132343637303134)
1-0:1.8.1(018856.351*kWh)
1-0:1.8.2(012856.725*kWh)
1-0:2.8.1(005963.866*kWh)
1-0:2.8.2(012674.532*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.020*kW)
1-0:2.7.0(00.052*kW)
0-0:96.7.21(00015)
0-0:96.7.9(00008)
1-0:99.97.0(6)(0-0:96.7.19)(221229144636W)(0000000793*s)(211125131827W)(0000014723*s)(210704021912S)(0000000717*s)(210704003443S)(0000005394*s)(190505022558S)(0000000425*s)(000101000013W)(2147483647*s)
1-0:32.32.0(00001)
1-0:52.32.0(00001)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.071*kW)
1-0:41.7.0(00.014*kW)
1-0:42.7.0(00.000*kW)
1-0:61.7.0(00.013*kW)
1-0:62.7.0(00.000*kW)
!46FC
```

bart1604, 2024-10-21:
Nou, ik ga hem gelijk weer sluiten.
Denk zal toch nog eens kijken en zie in home assistant wel 15 updates per seconden binnen krijgen. 
Dus ik krijg in de mqtt backlog van dsmr, zie het binnen 1minuut teruglopen van 5000 naar 0.
Probleem lijkt zichzelf opgelost te hebben.

dennissiemensma, 2024-10-21:
Zie ook #1999