dsmrreader/dsmr-reader #2032: Belgisch capaciteitstarief: hoe weet je of de berekende kwartierwaarde correct is?

Jeroen2000, 2024-10-28:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Eerder een bezorgdheid met ref. naar https://github.com/dsmrreader/dsmr-reader/issues/1084

Ik begrijp dat men voor de berekening bij de Belgische digitale meter (DM) rekening houdt met vaste kwartieren (9:00, 9:15, 9:30, ...) maar dat impliceert dat zowel de DSRM-reader SW als de DM gesyncroniseerd zijn qua klok? Nu is standaard NTP hier zeker nauwkeurig genoeg voor. Maar...

Wat als de klok in de DM bv. 5 minuten verkeerd staat of de klok op de host waar DSMR-reader draait verkeerd staat. Dan berekeken je het gemiddelde niet langer over dezelfde 900 seconden waardoor je wellicht andere kwartierpiekwaardes krijgt (tenzij de load landurig genoeg exact hetzelfde is). Mijn vraag is dus:  hoe wordt dit ondervangen en kan je nagaan of je de kwartierpiek gemeten door de DM overeenkomt met die berekend door DSMR-reader?

Alvast bedankt voor de toelichting en hopelijk is het een valide vraag:-)

### DSMR-reader version

5.10.2

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-10-28:
Er is geen garantie dat het 100% nauwkeurig is wanneer er geen meting is die precies aan het begin of einde van elke kwartier gemaakt wordt.

Dat heeft niet te maken met de klok van je hardware, want de tijdstippen staan in elke meting zelf, maar meer met de intervallen van de metingen. 
Dat zeggende hebben geven de latere generatie meters elke seconde een meting, dus een afwijking is verwaarloosbaar klein. Voor oudere meters die elke 10 seconden een meting geven is de afwijking iets meer, maar nog steeds minimaal (max 20 seconden op 900 seconden van een kwartier). 

Zie ook mijn eigen metingen destijds qua start/eindtijd (met een extra sleep van 10s): https://github.com/dsmrreader/dsmr-reader/issues/1084#issuecomment-1132207280

Overigens is DSMR-reader nooit leidend, maar meer een indicatie die vrij in de buurt komt van wat je energieleverancier of netbeheerder zelf meet. Het is bijvoorbeeld ook niet bedoeld om sluitend spiegelend te zijn, maar om een (redelijk nauwkeurige) benadering te geven van je metergegevens.

Jeroen2000, 2024-10-28:
@dennissiemensma Goed punt over de intervallen, ik heb een laaste gen. dus dat zit snor en ditto over de time stamp in de meting

Ik bedoel eerder een ernstiger type afwijking:

1. DM meet van (bijvoorbeeld) 9:20 tot 9:35. Van 9:35 tot 9:50. Van 9:50 tot 10:05,.... 
Redenen: iemand heeft de hoofdautomaat uitgezet en net om 9u20 weer aangeschakeld, stroompanne, ...
3. DSMR-reader meet met "vaste kwartieren" van 9:00 tot 09:15. Van 9u15 tot 9u30

In beide gevallen worden er telkens 900 seconden gemeten maar resulteert dit in andere kwartierpieken.
Is dat geen dingetje of meet de DM met vaste kwartieren?

dennissiemensma, 2024-10-28:
Ik kan je helaas niet vertellen hoe de slimme meter intern werkt. Voor de rest heb ik alles vooral gebaseerd op degene die destijds de feature heeft aangevraagd, die leek er vrij diep in te zitten: https://github.com/dsmrreader/dsmr-reader/issues/1084#issuecomment-1135114607

Voor de rest is het dus een benadering. De kwartierpieken staan ook in de telegrammen van de slimme meter, maar dat is een feature die er nog (lang) niet in zit: #1764

Dat zou het in theorie wel gelijktrekken.

closed 2024-11-07 by dennissiemensma