dsmrreader/dsmr-reader #2043: ?Traagheid na Home Assistant integratie MQTT

TommiieeSV, 2024-12-25:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [X] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [X] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [X] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [X] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Hallo,

Ik ben eergisteren begonnen met het spelen met Home Assistant. Ik zag dat er ook een [integratie](https://www.home-assistant.io/integrations/dsmr_reader/) is voor DSMR reader, dus ik ben daar mee gaan spelen.

Ik heb in Home Assistant een MQTT broker geïnstalleerd en deze ingesteld in DSMR Reader. Maar zodra ik deze zaken per de instructies enable, is DSMR reader niet meer vooruit te branden:

-     Day consumption: Split topic
-     Gas consumption: Split topic
-     Meter Statistics: Split topic
-     Quarter-hour peak consumption: Split topic
-     Telegram: Split topic

Het lijkt allemaal juist geconfigureerd, HA krijgt alles netjes binnen. Maar als ik DSMR Reader wil bekijken duurt het secondenlang voordat pagina's laden en bij 'About' krijg ik ook steeds meer meldingen dat hij e.e.a. niet verwerkt krijg.

Mijn setup is dat DSMR Reader in een Docker container op een Synology draait als API Server en de P1 poort is aangesloten op een RPi3 met API client.

Ik heb nu de MQTT integratie maar weer uitgezet, maar ik zou dit toch wel graag goed werkend krijgen. Graag hulp :)

### DSMR-reader version

5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

dennissiemensma, 2024-12-26:
Bedankt voor je melding. Zou je een debug info dump willen delen?
https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/debug-info-dump.html

Verder zou je in de log van het backend-proces moeten zien wat er allemaal gebeurt en wat er lang duurt:
https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

Je kunt voor de logs tijdelijk DEBUG-logging inschakelen, zodat je alles ziet. De exacte meldingen uit de About-pagina zijn ook handig om te delen.

Het zou kunnen zijn dat de MQTT-verwerking te traag is en dat dat alles ophoudt. Of dat er teveel MQTT-berichten verstuurd worden, die ophopen en de achterstand niet meer ingehaald wordt. 
Vooral de combinatie van meerdere split-topics kan zorgen voor een grote stroom aan berichten, afhankelijk van welke velden je allemaal ingeschakeld hebt.

TommiieeSV, 2024-12-26:
Dank voor de hulp. 

Ik heb de MQTT spullenboel weer even aangezet om verder te troubleshooten.

De debug info dump:
`DSMR-READER
    App | Python | Database                                                      v5.11 | v3.11.9 | postgresql
    BE sleep | DL sleep | Retention | Override                                     1.0s | 5.0s | 744h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                            1364997
    Consumption records electricity | gas (est.)                                                   63412 | -1

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 515 MB

`

De logs van het backend proces kan ik niet vinden. Mijn docker container heeft onder /var/log geen 'supervisor' folder. De datalogger op de RPi in de meterkast heeft die wel. Mijn docker container bevat onder /var/log een 4-tal folders: dsmr_backend, dsmr_datalogger, dsmr_webinterface en nginx. Alleen die laatste bevat logfiles, de 3 dsmr folders bevatten enkel lege bestanden met de namen 'current', 'lock' en 'state'.

Ik heb op de About page nog geen foutmeldingen, maar die komen dadelijk vanzelf terug. Zodra ik ze heb zal ik ze posten. Tot die tijd kunnen we misschien uitzoeken hoe ik aan de logs ga komen als ik geen supervisor folder heb onder /var/log.

dennissiemensma, 2024-12-30:
De logs waar ik op doel zijn overigens deze: https://github.com/xirixiz/dsmr-reader-docker/issues/338

Ik draai zelf nog geen Docker, maar je zou de logs van de container zelf kunnen bekijken. Als je ze niet kan vinden, dan wellicht met iets als `docker logs <container naam>`

TommiieeSV, 2024-12-30:
Ik heb een poging gedaan. Mijn default loglevel staat op ERROR. Ik heb deze eerst op WARNING gezet. Ik krijg dan dit:
```
2024/12/30 20:55:16	stdout	Current logging level set to "WARNING". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2024/12/30 20:55:13	stdout	Starting DSMR Reader - backend...
2024/12/30 20:55:11	stdout	2024-12-30 20:55:11,392 WARNING  dsmr_backend _check_restart_required          49 | Detected backend restart required, stopping process...
2024/12/30 20:55:11	stdout	2024-12-30 20:55:11,110 ERROR    persistent_clients run                              42 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2024/12/30 20:55:10	stdout	2024-12-30 20:55:10,964 WARNING  broker       signal_reconnect                146 | MQTT: Client no longer connected. Signaling restart to reconnect...
2024/12/30 20:55:10	stdout	2024-12-30 20:55:10,378 WARNING  broker       on_disconnect                   183 | MQTT: --- Unexpected disconnect, re-connecting...
2024/12/30 20:48:01	stdout	2024-12-30 20:48:01,220 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '19795' (4D53) does not match the telegram checksum '15858' (3DF2)
2024/12/30 20:46:29	stdout	2024-12-30 20:46:29,534 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '38649' (96F9) does not match the telegram checksum '52909' (CEAD)
2024/12/30 20:42:50	stdout	2024-12-30 20:42:50,988 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '38595' (96C3) does not match the telegram checksum '56582' (DD06)
2024/12/30 20:37:13	stdout	[2024-12-30 20:37:13 +0100] [19814] [INFO] Booting worker with pid: 19814
2024/12/30 20:37:12	stdout	[2024-12-30 20:37:12 +0100] [144] [INFO] Worker exiting (pid: 144)
2024/12/30 20:37:12	stdout	[2024-12-30 20:37:12 +0100] [144] [INFO] Autorestarting worker after current request.
2024/12/30 20:32:08	stdout	Current logging level set to "WARNING". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2024/12/30 20:32:04	stdout	Starting DSMR Reader - backend...
2024/12/30 20:32:01	stdout	2024-12-30 20:32:01,478 WARNING  dsmr_backend _check_restart_required          49 | Detected backend restart required, stopping process...
2024/12/30 20:32:01	stdout	2024-12-30 20:32:01,226 ERROR    persistent_clients run                              42 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
2024/12/30 20:32:00	stdout	2024-12-30 20:32:00,850 WARNING  broker       signal_reconnect                146 | MQTT: Client no longer connected. Signaling restart to reconnect...
2024/12/30 20:32:00	stdout	2024-12-30 20:32:00,080 WARNING  broker       on_disconnect                   183 | MQTT: --- Unexpected disconnect, re-connecting...
2024/12/30 20:27:42	stdout	2024-12-30 20:27:42,467 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '63657' (F8A9) does not match the telegram checksum '59657' (E909)
2024/12/30 20:21:23	stdout	2024-12-30 20:21:23,446 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '50932' (C6F4) does not match the telegram checksum '20874' (518A)
2024/12/30 20:16:52	stdout	2024-12-30 20:16:52,369 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '26904' (6918) does not match the telegram checksum '60088' (EAB8)
2024/12/30 20:12:48	stdout	2024-12-30 20:12:48,291 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '31764' (7C14) does not match the telegram checksum '12377' (3059)
2024/12/30 20:11:53	stdout	2024-12-30 20:11:53,314 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '48598' (BDD6) does not match the telegram checksum '33716' (83B4)
2024/12/30 20:11:42	stdout	2024-12-30 20:11:42,339 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Invalid telegram CRC. The calculated checksum '10987' (2AEB) does not match the telegram checksum '34542' (86EE)
2024/12/30 20:08:12	stdout	2024-12-30 20:08:12,781 WARNING  broker       on_disconnect                   183 | MQTT: --- Unexpected disconnect, re-connecting...
2024/12/30 20:08:12	stdout	2024-12-30 20:08:12,714 WARNING  broker       run                              90 | MQTT: Dropped 1 message(s) from queue due to limit
```
Ik heb ook DEBUG logging geprobeerd, maar die krijg ik hier moeilijk neergezet ivm max characters.

Het werkt dus allemaal wel, maar het is dus een stuk trager dan zonder MQTT. Ik heb nu één alert:

`Process behind schedule: Calculate quarter hour electricity peaks   (2 days, 8 hours ago)`

dennissiemensma, 2024-12-31:
Voor de debug-logging is het belangrijk dat je kijkt of je ziet of het versturen/verwerken van MQTT-berichten traag gaat. 

[Bij dit voorbeeld](https://github.com/xirixiz/dsmr-reader-docker/issues/338#issuecomment-1633088700) gaat het bijvoorbeeld redelijk vlot, elk bericht kost telkens maar paar ms qua tijd, maar de vraag is of je niet meer berichten probeert te versturen dan je MQTT-broker aan kan.

Je datalogger sleep staat op 5 seconden, dus DSMR-reader heeft 5 seconden om alle MQTT-berichten voor alle topics voor elk telegram (en wat je ook allemaal aan hebt staan) te versturen. Als dat niet in 5 seconden lukt, raakt die vanzelf steeds verder achter en hoopt het op.

Het versturen van de MQTT-berichten gebeurt in hetzelfde proces als de rest, wat een `Proces loopt achter op schema:` kan veroorzaken. Maar dat blijven aannames.

TommiieeSV, 2025-01-03:
Dank voor je reactie. Ik heb wat dagen moeten wachten omdat schijven heb vervangen in de Syno, waardoor de RAID moest rebuilden. In die tijd heb ik DSMR Reader maar even uitgeschakeld.

Ik heb gekeken in de DEBUG logging die ik bewaard had, het lijkt er toch op dat de MQTT berichten snel verwerkt worden:

```
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,995 DEBUG    messages     queue_message                    47 | MQTT: Queued message for dsmr/meter-stats/voltage_sag_count_l1: 7
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,808 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090904) from queue
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,808 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 426)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,807 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090904) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,807 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090904) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,807 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 426)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,807 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 426)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,806 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090904) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,806 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m426), 'b'dsmr/reading/phase_voltage_l1'', ... (5 bytes)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,805 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090904) for dsmr/reading/phase_voltage_l1: 230.0
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,561 DEBUG    messages     queue_message                    47 | MQTT: Queued message for dsmr/meter-stats/long_power_failure_count: 16
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,408 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090903) from queue
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,408 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 425)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,407 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090903) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,407 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090903) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,407 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 425)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,407 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 425)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,406 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090903) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,406 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m425), 'b'dsmr/reading/phase_currently_returned_l3'', ... (5 bytes)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,405 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090903) for dsmr/reading/phase_currently_returned_l3: 0.000
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,228 DEBUG    messages     queue_message                    47 | MQTT: Queued message for dsmr/meter-stats/power_failure_count: 28
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,075 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090902) from queue
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,075 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 424)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,074 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090902) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,074 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090902) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,074 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 424)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,074 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 424)
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,073 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090902) to be marked published by broker
2024/12/30 21:12:38	stdout	2024-12-30 21:12:38,072 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m424), 'b'dsmr/reading/phase_currently_returned_l2'', ... (5 bytes)
[... cut ...]
2024/12/30 21:12:35	stdout	2024-12-30 21:12:35,472 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090898) for dsmr/reading/phase_currently_delivered_l3: 0.000
2024/12/30 21:12:35	stdout	2024-12-30 21:12:35,142 DEBUG    messages     queue_message                    47 | MQTT: Queued message for dsmr/reading/phase_power_current_l2: 0
```

Ik heb MQTT nu weer even uitgezet en alle berichten in de queue verwijderd. Straks ga ik even spelen met de datalogger sleep instellingen. Ik merk dat de traagheid vooral inzet als ik de Telegram split topic enable.

TommiieeSV, 2025-01-03:
Ik heb zojuist uitgebreid getest.

De volgende topics werken prima wanneer enabled:
- Day consumption
- Quarter hour peak

DSMR Reader blijft snel en pagina's openen instant.

De problematiek begint bij deze twee:
- Meter statistics
- Telegram

De problemen beginnen al als slechts één van deze enabled is. Het laden van pagina's (zowel front-end als back-end) gaat dan richting de 5 à 10 seconden.

Ik heb getest met Meter Statistics split topic. Als ik deze enable, dan zie ik de queue oplopen. Als ik het topic weer disable, is de queue binnen no-time leeg. Ik heb daarop de datalogger sleep verhoogt in stappen, maar uiteindelijk naar 99. Hij weet dan de queue in principe leeg te vegen, maar:
- Er blijven altijd 6 berichten staan (of dat lijkt zo)
- De traagheid bestaat al zodra het topic enabled is, ongeacht hoeveel berichten er in de queue staan. Behalve als de max van 5000 bereikt wordt, dan crasht heel DSMR-reader en moet ik de container opnieuw starten.

Hetzelfde gedrag zie ik bij Telegram split topic. Ongeacht hoeveel berichten er in de queue staan of op welke value datalogger sleep staat, zodra deze is aangevinkt, duurt het laden van pagina's 5 à 10 seconden.

Ik zie dus in feite 2 problemen:
- Als ik terugkijk naar de startpost en ik enable alle genoemde topics en ik pas **niet** de datalogger sleep aan, dan weet hij de queue niet te legen. Uiteindelijk crasht DSMR reader. Ik kan het niet helemaal matchen met de debug logging, daar lijkt alles toch aardig snel verwerkt te worden, al valt een ~0,3 seconde vertraging na een queue_message wel op.
- De traagheid van de interface (het laden van pagina's) lijkt niet gerelateerd aan queue size, maar wordt direct geïmpact zodra Telegram en/of Meter statistics split topic wordt enabled.

dennissiemensma, 2025-01-03:
Er zijn meerdere dingen die je kunt proberen:
- Minder berichten versturen door ofwel de gebundelde JSON topics te gebruiken of minder topics
- BE-sleep van 1 seconde naar 0,5 of 0 zetten
- VacuumDB doen zodat Postgres even alles weer soort van defragmenteert

----

De split-topics sturen een bericht voor elke regel die je aan hebt staan. Hier regels weghalen scheelt per regel 1 bericht per keer:
[image]

Dat geldt voor elke split-topic die je aan hebt staan.

----

VacuumDB kan met: 
- https://github.com/xirixiz/dsmr-reader-docker?tab=readme-ov-file#db-related-defaults-are-shown-as-value
- https://github.com/xirixiz/dsmr-reader-docker?tab=readme-ov-file#dsmr-reader---database-cleanupvacuum

dennissiemensma, 2025-01-03:
Is de debug log de rauwe debug log? Of doe je een grep op `MQTT`?

Je zou in principe alles moeten zien wat er gebeurt, ook niet-MQTT-zaken, mits die voorbij komen. De korte vertraging van 0,1s - 0,3s kan meerdere oorzaken hebben:
- Er gebeurt iets anders wat niet in de log staat (verwerking)
- De DB is traag (vacuumDB kan helpen, al zou dat niet regelmatig nodig moeten zijn, puur bij het verwijderen van een hoop data in korte tijd)

Ik vermoed dat het het eerste is, gezien het queuen van berichten en het versturen losse acties zouden moeten zijn en ze in de logs als 1 actie lijken.

TommiieeSV, 2025-01-05:
De debug log was een snippet, ivm max aantal tekens hier. Hier is een wat langere snippet, tussen 2 telegram receives in:

```
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,182 DEBUG    datalogger   telegram_to_reading              81 | Received telegram:
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,043 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090875) from queue
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,042 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 397)
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,042 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090875) to be marked published by broker
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,042 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090875) to be marked published by broker
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,042 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 397)
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,041 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 397)
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,041 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090875) to be marked published by broker
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,040 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m397), 'b'dsmr/reading/phase_power_current_l2'', ... (1 bytes)
2024/12/30 21:12:29	stdout	2024-12-30 21:12:29,040 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090875) for dsmr/reading/phase_power_current_l2: 0
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,926 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090874) from queue
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,926 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 396)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,922 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090874) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,922 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090874) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,921 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 396)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,921 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 396)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,918 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090874) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,917 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m396), 'b'dsmr/reading/phase_power_current_l1'', ... (1 bytes)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,917 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090874) for dsmr/reading/phase_power_current_l1: 4
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,721 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090873) from queue
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,720 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 395)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,720 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090873) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,719 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090873) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,719 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBREL (Mid: 395)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,719 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBREC (Mid: 395)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,718 DEBUG    broker       run                             127 | MQTT: Waiting for message (#1090873) to be marked published by broker
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,718 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m395), 'b'dsmr/reading/phase_voltage_l3'', ... (5 bytes)
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,718 DEBUG    broker       run                             103 | MQTT: Publishing queued message (#1090873) for dsmr/reading/phase_voltage_l3: 231.0
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,652 DEBUG    broker       run                             140 | MQTT: Deleting published message (#1090872) from queue
2024/12/30 21:12:28	stdout	2024-12-30 21:12:28,652 DEBUG    broker       on_log                          194 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 394)
[... cut ...]
- De BE-sleep naar 0 zitten leek lichte verbetering te brengen, maar het is alsnog flink te traag.

Het minderen van berichten zie ik niet als oplossing maar als workaround. Liefst blijf ik trouw aan de handleiding van de HA integratie.

Wellicht dat overblijft dat de DB en/of DSMR-Reader te traag is, maar valt het pas op zodra MQTT werk moet gaan verzetten. Enige oplossing die ik daarvoor kan bedenken is SSD's in de NAS plaatsen (duur) of wellicht de DSMR-DB op de Pi van de datalogger gaan draaien.

dennissiemensma, 2025-01-05:
Heb je echt alle velden nodig? Want dan is het efficienter om het per type te groeperen met JSON, vooral die van de meter statistieken en telegrammen.

Als je alles hebt aanstaan kom ik uit op een kleine 60 (!) topics. Hoewel het technisch moet kunnen, is het ook een kwestie van balanceren wat in de praktijk werkt. En dat verschilt per set-up.

En ik zie bijvoorbeeld dat de gaswaardes leeg zijn, dus die zou ik er dan vooral uithalen. En mocht je bijvoorbeeld geen teruglevering hebben, vooral uitzetten.

```ini
electricity1 = dsmr/day-consumption/electricity1
electricity2 = dsmr/day-consumption/electricity2
electricity1_returned = dsmr/day-consumption/electricity1_returned
electricity2_returned = dsmr/day-consumption/electricity2_returned
electricity_merged = dsmr/day-consumption/electricity_merged
electricity_returned_merged = dsmr/day-consumption/electricity_returned_merged
electricity1_cost = dsmr/day-consumption/electricity1_cost
electricity2_cost = dsmr/day-consumption/electricity2_cost
electricity_cost_merged = dsmr/day-consumption/electricity_cost_merged
gas = dsmr/day-consumption/gas
gas_cost = dsmr/day-consumption/gas_cost
total_cost = dsmr/day-consumption/total_cost
fixed_cost = dsmr/day-consumption/fixed_cost
energy_supplier_price_electricity_delivered_1 = dsmr/day-consumption/energy_supplier_price_electricity_delivered_1
energy_supplier_price_electricity_delivered_2 = dsmr/day-consumption/energy_supplier_price_electricity_delivered_2
energy_supplier_price_electricity_returned_1 = dsmr/day-consumption/energy_supplier_price_electricity_returned_1
energy_supplier_price_electricity_returned_2 = dsmr/day-consumption/energy_supplier_price_electricity_returned_2
energy_supplier_price_gas = dsmr/day-consumption/energy_supplier_price_gas
read_at = dsmr/consumption/gas/read_at
delivered = dsmr/consumption/gas/delivered
currently_delivered = dsmr/consumption/gas/currently_delivered
dsmr_version = dsmr/meter-stats/dsmr_version
electricity_tariff = dsmr/meter-stats/electricity_tariff
power_failure_count = dsmr/meter-stats/power_failure_count
long_power_failure_count = dsmr/meter-stats/long_power_failure_count
voltage_sag_count_l1 = dsmr/meter-stats/voltage_sag_count_l1
voltage_sag_count_l2 = dsmr/meter-stats/voltage_sag_count_l2
voltage_sag_count_l3 = dsmr/meter-stats/voltage_sag_count_l3
voltage_swell_count_l1 = dsmr/meter-stats/voltage_swell_count_l1
voltage_swell_count_l2 = dsmr/meter-stats/voltage_swell_count_l2
voltage_swell_count_l3 = dsmr/meter-stats/voltage_swell_count_l3
rejected_telegrams = dsmr/meter-stats/rejected_telegrams
read_at_start = dsmr/consumption/quarter-hour-peak-electricity/read_at_start
read_at_end = dsmr/consumption/quarter-hour-peak-electricity/read_at_end
average_delivered = dsmr/consumption/quarter-hour-peak-electricity/average_delivered
id = dsmr/reading/id
timestamp = dsmr/reading/timestamp
electricity_delivered_1 = dsmr/reading/electricity_delivered_1
electricity_returned_1 = dsmr/reading/electricity_returned_1
electricity_delivered_2 = dsmr/reading/electricity_delivered_2
electricity_returned_2 = dsmr/reading/electricity_returned_2
electricity_currently_delivered = dsmr/reading/electricity_currently_delivered
electricity_currently_returned = dsmr/reading/electricity_currently_returned
phase_currently_delivered_l1 = dsmr/reading/phase_currently_delivered_l1
phase_currently_delivered_l2 = dsmr/reading/phase_currently_delivered_l2
phase_currently_delivered_l3 = dsmr/reading/phase_currently_delivered_l3
phase_currently_returned_l1 = dsmr/reading/phase_currently_returned_l1
phase_currently_returned_l2 = dsmr/reading/phase_currently_returned_l2
phase_currently_returned_l3 = dsmr/reading/phase_currently_returned_l3
extra_device_timestamp = dsmr/reading/extra_device_timestamp
extra_device_delivered = dsmr/reading/extra_device_delivered
phase_voltage_l1 = dsmr/reading/phase_voltage_l1
phase_voltage_l2 = dsmr/reading/phase_voltage_l2
phase_voltage_l3 = dsmr/reading/phase_voltage_l3
phase_power_current_l1 = dsmr/reading/phase_power_current_l1
phase_power_current_l2 = dsmr/reading/phase_power_current_l2
phase_power_current_l3 = dsmr/reading/phase_power_current_l3
```

dennissiemensma, 2025-01-05:
Lost de `vacuumdb` wel de melding op van de tabelgrootte?

```
> POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
>     public.dsmr_datalogger_dsmrreading                                                                 515 MB
```

dennissiemensma, 2025-01-05:
Ik zie dat het gat na elke `queue_message` aardig varieert. Soms is het slechts 0,1s, wat nog acceptabel is.

Als je nog dieper wilt debuggen, zul je denk ik op de database de query log moeten aanzetten en kijken of je daar trage queries tussen ziet staan. De query log kan invloed hebben op de performance, maar het is toch maar even tijdelijk.
- https://stackoverflow.com/a/58806511

In de meeste gevallen is de grootte van de database i.c.m. hardware de bottleneck, maar gezien je 1,3 miljoen metingen hebt staat icm retentie, zijn dat geen schokkende cijfers.

dennissiemensma, 2025-01-06:
Je kunt trouwens deze instelling ook nog proberen:

https://dsmr-reader.readthedocs.io/en/v5/reference/env-settings.html#dsmrreader-mqtt-max-cache-timeout

TommiieeSV, 2025-01-07:
> Je kunt trouwens deze instelling ook nog proberen:
> 
> https://dsmr-reader.readthedocs.io/en/v5/reference/env-settings.html#dsmrreader-mqtt-max-cache-timeout

Dit lijkt goed te werken! Met deze timeout enabled heb ik met MeterStatistics getest en de interface bleef snel. Met het enablen van Telegrams kwam de traagheid terug, maar wel in veel mindere mate. Ik vind dit aardig acceptabel, al ga ik wel kijken of ik het nog gefinetuned krijg.

Voor zover ik weet heb ik al alles van gas uitstaan, want ik heb een gasloze woning. Ik kom ook geen melding tegen van een tabelgrootte > 500MB. Waar heb je die gezien? Als ik naar de size van de hele postgresql folder kijk die bij de container hoort dan is die hele folder zelfs < 500 MB.

We zijn al een heel stuk verder, goede tip!

dennissiemensma, 2025-01-07:
De melding van de grootte van de database staat in de debug dump: https://github.com/dsmrreader/dsmr-reader/issues/2043#issuecomment-2562781614

dennissiemensma, 2025-01-07:
En nog voor de volledigheid de info over die caching. In de toekomstige docs is het iets anders verwoord:

[image]

TommiieeSV, 2025-01-09:
> De melding van de grootte van de database staat in de debug dump: [#2043 (comment)](https://github.com/dsmrreader/dsmr-reader/issues/2043#issuecomment-2562781614)

Sorry, die had ik gemist. Die melding is nu verdwenen inderdaad.

So far so good. Nog geen rode meldingen in de About. Performance is "okay", niet instant, maar niet traag genoeg om echt last van te hebben.

Thanks!

dennissiemensma, 2025-01-09:
Mocht je ooit nog verder willen optimaliseren, dan zou je de DB logs erbij kunnen pakken om te zien of daar iets traag is

https://github.com/dsmrreader/dsmr-reader/issues/2043#issuecomment-2571737699

closed 2025-01-09 by dennissiemensma