dsmrreader/dsmr-reader #2048: Geen gas zichtbaar in DSMRReader - Fluvius - Docker container

puijken, 2025-01-29:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Ik heb bij een kameraard een installatie gedaan van DSMRReader na het zelf al enkele jaren te gebruiken. Hij had reeds een HomeWizard P1 meter aangekocht en via het script heb ik uiteindelijk de telegram berichten van de meter in DSMRReader gekregen. 

Echter blijkt dat er geen gas zichtbaar is .. en ik zie niet onmiddelijk waarom. Ik heb de issues even doorzocht en zag dat Fluvius soms wat lastig kan doen en ik heb ondertussen wat met de instellingen gespeeld om een kanaal te kiezen (naast de automatische setting) maar niks lijkt uiteindelijk gas te registreren. 

Als ik de issues goed bekeken heb lijkt het ook dat (ondanks dat ik gebruik maak van een script voor de HomeWizard P1) het telegram wel correct binnenkomt, echter begrijp ik niet onmiddelijk waaorm het niet verwerkt wordt.

Ik zie ook de waarde omhoog gaan voor:
**0-1:24.2.3(250129124652W)(03454.105*m3)**

Uit de voorgaande issues begreep ik da dit "kanaal 1" zou moeten zijn, maar ook dan wordt er niks verwerkt.

### DSMR-reader version

5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script
/FLU5\253769484_A

0-0:96.1.4(50217)
0-0:96.1.1(3153414733313030303738333838)
0-0:1.0.0(250129124903W)
1-0:1.8.1(003005.126*kWh)
1-0:1.8.2(002964.764*kWh)
1-0:2.8.1(004662.658*kWh)
1-0:2.8.2(001994.581*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.000*kW)
1-0:1.6.0(250116173000W)(02.340*kW)
0-0:98.1.0(13)(1-0:1.6.0)(1-0:1.6.0)(240101000000W)(231218144500W)(03.196*kW)(240201000000W)(240125201500W)(02.507*kW)(240301000000W)(240208173000W)(02.524*kW)(240401000000S)(240304064500W)(01.830*kW)(240501000000S)(240412180000S)(00.607*kW)(240601000000S)(240503094500S)(00.540*kW)(240701000000S)(240629190000S)(00.849*kW)(240801000000S)(240713140000S)(01.003*kW)(240901000000S)(240810111500S)(00.469*kW)(241001000000S)(240903080000S)(00.105*kW)(241101000000W)(241021124500S)(01.672*kW)(241201000000W)(241108173000W)(01.734*kW)(250101000000W)(241212173000W)(02.374*kW)
1-0:1.7.0(00.004*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.015*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.010*kW)
1-0:32.7.0(229.6*V)
1-0:52.7.0(000.0*V)
1-0:72.7.0(230.4*V)
1-0:31.7.0(001.16*A)
1-0:51.7.0(001.31*A)
1-0:71.7.0(001.04*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313139303331353638)
0-1:24.4.0(1)
0-1:24.2.3(250129124652W)(03454.105*m3)
!1D36
```

dennissiemensma, 2025-02-02:
Zou je een debug info dump willen delen? 
https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/debug-info-dump.html

Verder zou je DEBUG-logging aan kunnen zetten en kijken of daar nog meldingen uitkomen qua verwerking:
https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

puijken, 2025-02-04:
Jazeker, hierbij de gevraagde gegevens. Ik zie dat de gaswaarde omhoog gaat van 

0-1:24.2.3(250204204008W)(03476.125*m3)
naar
0-1:24.2.3(250204204606W)(03476.223*m3)
en naar
0-1:24.2.3(250204205000W)(03476.296*m3)

Echter zie ik in de debug log niet onmiddelijk waarom dit niet verwerkt wordt. De debug info laat wel iets interessant zien (geen gas consumption):

`DSMR-READER
    App | Python | Database                                                     v5.11 | v3.11.11 | postgresql
    BE sleep | DL sleep | Retention | Override                                     5.0s | 5.0s | 744h | False
    Latest telegram version read | Parser settings                                               "None" | "4"

DATA
    Telegrams total (est.)                                                                             145005
    Consumption records electricity | gas (est.)                                                    11716 | -
`

[dsmrreader_debug.txt](https://github.com/user-attachments/files/18662786/dsmrreader_debug.txt)

puijken, 2025-02-04:
Ok, feeling stupid ... het is opgelost door natuurlijk de goede instelling te kiezen, door alle instellingen met kanalen heen los over deze setting gekeken.

[image]

Alhoewel ik er ook wel over gelezen had, door dit op Belgie te zetten werd de laatste gaswaarde onmiddelijk zichtbaar.
Stom! In elk geval bedankt voor de moeite!

closed 2025-02-04 by puijken