dsmrreader/dsmr-reader #2052: Meter (BE) vervangen, jaartotaal negatief

p4r4ll4x, 2025-02-17:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Mijn digitale meter is onlangs vervangen.
De dagwaarden zijn correct behalve op de dag van vervanging.

Echter het maand- en jaartotaal zijn volledig verkeerd.
Wat is de procedure om deze waarden te corrigeren?

[image]

I find the following in the docs: https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/smart-meter-replacement.html
But i don't seem to have invalid data.

### DSMR-reader version

v5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-02-17:
Je zult (ook) de dag-statistieken aan moeten passen van die dag. De data uit je screenshot zijn de uur-statistieken.

dennissiemensma, 2025-02-17:
- `/admin/dsmr_stats/daystatistics/`

p4r4ll4x, 2025-02-18:
> * `/admin/dsmr_stats/daystatistics/`

Bedankt, ik heb daar het dagtotaal realistisch gemaakt en het jaartotaal werd meteen aangepast.
Bedankt voor de info!