dsmrreader/dsmr-reader #2055: How to completely disable any logging for Remote Datalogger ?

nero355, 2025-03-12:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

First of all :

Thank you for creating this totally AWESOME!!! piece of software 👍🏻 👍🏻 👍🏻 👍🏻 👍🏻 

But I need a bit of help : 

Is it possible to disable ALL logging for the Remote Datalogger : https://dsmr-reader.readthedocs.io/en/v5/how-to/installation/remote-datalogger.html#supervisor
So basically this piece of code/config : /etc/supervisor/conf.d/dsmr_remote_datalogger.conf
Which contains : 
`stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=10MB
stdout_logfile_backups=3`
I have tried setting it to 1MB and 1 file but it seems it then just overwrites itself ?!

Can I completely remove the stdout_log lines to avoid any logging ?

The reason I am asking this : 
I want to avoid "damage" to the microSDXC card in the Raspberry Pi 3B that servers as the Remote Datalogger whenever I reboot the actual DSMR Server or whenever I am upgrading/changing things on my network devices 😬

### DSMR-reader version

5.11

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-03-16:
Je kunt proberen om de drie Supervisor configs aan te passen van:

```
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
```

Naar

```
stdout_logfile=NONE
```

En daarna de wijziging toepassen met:
```
sudo supervisorctl reread
sudo supervisorctl update
```

https://supervisord.org/configuration.html#program-x-section-settings

[image]

closed 2025-03-16 by dennissiemensma

nero355, 2026-02-01:
Ik heb bovenstaande nou al een tijdje draaien, maar het heeft geen effect : Ik krijg gewoon een bestand met een door SuperVisor gegenereerde naam ?!

Dit is mijn config file : 
```
root@raspberrypi:/etc/supervisor/conf.d# cat dsmr_remote_datalogger.conf 
[program:dsmr_remote_datalogger]
command=/home/dsmr/.venv/bin/python3 -u /home/dsmr/dsmr_datalogger_api_client.py
pidfile=/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=20
redirect_stderr=false
stdout_logfile=NONE
```

En ik krijg alsnog zoiets : dsmr_remote_datalogger-stderr---supervisor-a6xjz1rt.log
Van maar liefst 50 MB aan logging !!

Tevens had ik een tijdje last van HTTP 500 Errors volgens dat bestand, maar dat lijkt nu over te zijn, want na de laatste keer het bestand verwijderd te hebben is er niks meer erbij gekomen gelukkig!

Ik blijf het een beetje in de gaten houden dus als je nog TIPS! hebt dan hoor ik die graag 😉