dsmrreader/dsmr-reader #2058: Datalogger process sleep >0.5 geeft usbip errors

ualex73, 2025-03-22:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

Mijn iowait is een beetje hoog, dus ik wilde het verlagen door de waarde "Datalogger process sleep" van 0.5 seconden naar 2-5 seconden te veranderen. Dit gaf mijn een lagere iowait, maar nu krijg ik andere problemen.

Mijn DSMR-Reader setup is als volgt:
- Ik draai een VirtualMachine op ProxMox
- In de VM draai ik Docker
- De serieel poort zet ik door via "usbip" (standaard onderdeel van Debian/Linux), want de Proxmox serieel poort doorgeven is minder stabiel

Dit draait met 0.5 seconden goed, maar zodra ik het hoger zet, zie ik continue op mijn Proxmox de volgende melding (3-7 is de USB poort van P1):
[Sat Mar 22 10:28:10 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:12 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:12 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:14 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:16 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:16 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:18 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()
[Sat Mar 22 10:28:18 2025] usbip-host 3-7: unlinked by a call to usb_unlink_urb()

Na een tijdje gaat deze usbip P1 plat (reageert niet meer) in mijn VM. Ik heb al gepoogd om uit te vogelen hoe de DSMR-Reader de serieel poort gebruikt, maar niet 1-2-3 kunnen vinden.
Het werkt wel met 0.5 seconden, maar wanneer hoger, dan doet DSMR-Reader iets "raars" (misschien elke keer serieel poort openen en sluiten?).

### DSMR-reader version

v5.11.0

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-03-22:
DSMR-reader maakt bij hogere sleeps telkens een nieuwe verbinding, de grens staat op 1 seconde. Je kunt die aanpassen (hoger zetten) met deze envvar:  DSMRREADER_REMOTE_DATALOGGER_MIN_SLEEP_FOR_RECONNECT https://github.com/dsmrreader/dsmr-reader/blob/6954590a1b82fe76ae9cec44397e57159fa197af/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py#L143

ualex73, 2025-03-22:
Dennis dank je voor het snelle antwoord, dus ik heb het gelijk geprobeerd. Ik heb de volgende env toegevoegd:
DSMRREADER_REMOTE_DATALOGGER_MIN_SLEEP_FOR_RECONNECT=6

Ik heb "Datalogger process sleep" op 2 en 4 seconden geprobeerd, maar de meldingen komen weer uit de usbip logging ... dus het lijkt het niet op te lossen.

dennissiemensma, 2025-03-23:
Je zou kunnen proberen de datalogger sleep op 0.9 en 1.1 seconde te zetten en dan te kijken of je verschil ziet in meldingen. Wellicht pakt de datalogger die limiet niet goed en blijft die op 1s staan.

dennissiemensma, 2025-03-24:
Welke datalogger sleep gebruik je trouwens? Via interface of envvars? Want dit werkt alleen met de remote datalogger en die kijkt naar de envvar DSMRREADER_REMOTE_DATALOGGER_SLEEP. Gezien je verschil ziet in gedrag, doet je sleep aanpassing wel iets (blijkbaar), maar goed om even te checken.

ualex73, 2025-03-24:
Ik heb het volgende in mijn docker-compose staan:
    environment:
      - DSMRREADER_SUPPRESS_STORAGE_SIZE_WARNINGS=True
      - DSMRREADER_REMOTE_DATALOGGER_MIN_SLEEP_FOR_RECONNECT=6

De "Datalogger process sleep" stel ik in via de GUI.

dennissiemensma, 2025-03-24:
Probeer het eens (ook) via DSMRREADER_REMOTE_DATALOGGER_SLEEP, wellicht pakt die hem dan wel.

ualex73, 2025-03-27:
Ook geprobeerd, maar jammer genoeg geen verschil in gedrag.

closed 2025-08-04 by dennissiemensma