dsmrreader/dsmr-reader #2060: Netto verbruik / teruglevering toevoegen

bart1604, 2025-04-07:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Ik heb een aantal onderwerpen gevonden omtrent dit onderwerk maar wil er toch nog een vraag over stellen.
Ik zou graag in het dashboard kunnen zien, en ook willen exporteren dmv MQTT wat het netto aantal watt is wat teruggeleverd wordt of verbruikt wordt.

De reden dat ik dit nu extra belangrijk ben gaan vinden is dat, ondanks dat ik de gegevens exporteer naar HomE Assistant. Ik vooral DSMR gebruik voor het (terug)kijken van verbruik. Nu ik van SolarEdge ook een Accu pakket heb aangeschaft, ik nu niet heel veel meer heb aan de data die DSMR mij geeft.

De omvormer kan uit de accu alleen op alle 3 de fases hetzelfde terugleveren, hij kan niet kiezen om op 1 fase te leveren. Dit resulteert in het volgende.

Als het verbruik het volgende is (even theoretisch)
fase 1 1000w gebruik
fase 2 0w gebruik
fase 3 0w gebruik.

Dan doet de accu hetvolgende
fase 1 333w terugleveren -> resulteerd in een gebruik van 666w
fase 2 333w terugleveren -> resulteerd in een teruggave van 333w
fase 3 333w terugleveren -> resulteerd in een teruggave van 333w

En omdat de meter zelf saldeert komt de meter op 0.

Om deze reden wordt het voor mij veel interresanter om te weten dat mij verbruik op dit moment 0 is. Tussen april en oktober (vanwege de panelen en accu) is mijn verbruik bijna heel de tijd 0. Ik kan nu met de data die ik heb en het archief niet achterhalen of ik 0 verbruik heb gehad of niet.

Ik hoor graag of dit een optie is. Wellicht een optie om dit te verwerken in de settings? Zodat men zelf kan kiezen.

### DSMR-reader version

4

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-04-07:
Bedankt voor je verzoek. Ik weet niet of dat uit de meter te halen is, want DSMR-reader saldeert zelf ook niet.

Wellicht kun je het uit MQTT topics halen voor de fasen, want meer informatie dan dat heeft DSMR-reader ook niet. En specifieke persoonlijke checks/situaties zijn beter te automatiseren via die weg, zodat DSMR-reader vrij neutraal blijft qua weergave van de meter.

dennissiemensma, 2025-04-21:
Ik denk dat wat je zoekt niet in DSMR-reader staat, ook omdat het wat misleidend kan zijn voor andere gebruikers, gezien het puur een reflectie is van de meter en niet een eigen netto. 

Als je per dag of uur het verbruik en teruglevering netto wil weten, kun je proberen om de CSV-export te gebruiken, dan kun je in Excel vrij makkelijk  de som maken die je zoekt. Ook in het verleden.

closed 2025-04-21 by dennissiemensma

opvos, 2025-04-23:
HA kent op dit moment inderdaad geen 'substract' functie voor sensorwaarden ;)
Binnen HA is het mogelijk een samengestelde sensor aan te maken, die de interne saldering van de DSMR meter berekent.

Mini Howto:
- Zoek eerst de IDs op uit van de huidige sensors voor levering en productie. Bijvoorbeeld: 'sensor.electricity_meter_power_consumption' en 'sensor.electricity_meter_power_production' 

- Maak een samengestelde sensor via: Settings > Devices > Helpers > Create Helper
Type: Template > Template a sensor >
Name: Electricity Meter NetPower
State template: 
`{{states('sensor.electricity_meter_power_consumption') | float(0) - states('sensor.electricity_meter_power_production') | float(0)}}`
Unit: kW
Device Class: Power
State Class: Measurement
Device: Electricity Meter
Display precision: Default
Entity ID: sensor.netpower

- De nieuwe sensor is vervolgens weer te gebruiken binnen 'Automation' en Lovelace