dsmrreader/dsmr-reader #2061: Probleem met actieve P1 splitter (HomeWizard)

henkbb36org, 2025-05-01:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Ik heb de actieve P1 splitter van HomeWizard in gebruik genomen omdat de HomeWizard P1 dongle met mijn HomeWizard thuisbatterij moet kunnen samenwerken. De HomeWizard P1 dongle werkt goed achter de splitter maar de serial2USB kabel naar mijn Linux server met dsmr-reader werkt niet goed wanneer de splitter er tussen zit. Alleen wanneer de serial2USB kabel rechtstreeks is aangesloten op de P1 werkt dsmr-reader dus wel.
Iemand hier ook tegenaan gelopen? Iemand ideeën hoe dit op te lossen?

### DSMR-reader version

5

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

henkbb36org, 2025-05-22:
Ik denk niet dat de HomeWizard splitter ser2net gebruikt omdat het geen netwerk device is. Ik heb de data nog met een programmaatje uitgelezen en zag dat deze er niet goed uitzag via de splitter, rechtstreeks aangesloten wel. Net alsof de baudrate niet goed was maar ik heb een aantal baudrates uitgeprobeerd, dat was het niet. Ik gebruik nu de API van de HomeWizard P1 meter daar heb ik voldoende aan. Heb jarenlang naar grote tevredenheid de DSMR-reader gebruikt, dikke complimenten voor jullie werk!

dennissiemensma, 2025-05-24:
Fijn om te horen! 😄

closed 2025-05-24 by dennissiemensma