dsmrreader/dsmr-reader #2063: MQTT werkt niet meer; geeft altijd maar één keer dezelfde waarde

cranberryNL, 2025-05-10:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Report a bug

### Description

DSMR werkt hier inmiddels al 6 jaar heel stabiel. Zojuist kwam ik er echter achter dat het rapporteren naar MQTT niet meer werkt.
Ik krijg eenmalig een bericht van DSMR als ik bijv. met MQTT Explorer 'luister' en dat is ook altijd dezelfde waarde.
Ik draai DSMR 5.11 en heb al een keer opnieuw een './deploy.sh' gedaan.

Als ik in de dsmr_datalogger.log kijk, zie ik heel veel van deze meldingen:
2025-05-10 21:48:52,816 ERROR    apps         _on_dsmrreading_created_signal  154 | publish_split_topic_dsmr_reading() failed: integer out of range

Ook de dsmr_backend.log geeft fouten:

2025-05-10 21:54:14,697 ERROR    apps         _on_electricity_consumption_created_signal   94 | publish_day_consumption() failed: integer out of range

2025-05-10 21:54:14,722 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: integer out of range

2025-05-10 21:55:15,854 ERROR    apps         _on_electricity_consumption_created_signal   94 | publish_day_consumption() failed: integer out of range

2025-05-10 21:55:15,882 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: integer out of range

2025-05-10 21:56:01,825 ERROR    apps         _on_electricity_consumption_created_signal   94 | publish_day_consumption() failed: integer out of range

2025-05-10 21:56:01,854 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: integer out of range

2025-05-10 21:57:02,984 ERROR    apps         _on_electricity_consumption_created_signal   94 | publish_day_consumption() failed: integer out of range

2025-05-10 21:57:03,010 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: integer out of range

Via de webinterface/live grafieken zie ik wel actuele, correcte data die iedere paar seconden wordt bijgewerkt.

Kan het zijn dat een waarde van mijn meter te groot is?
Hierbij een recent telegram:

1-3:0.2.8(50)
0-0:1.0.0(250510215942S)
0-0:96.1.1(4530303334303037343732373331383139)
1-0:1.8.1(010823.715*kWh)
1-0:1.8.2(012200.300*kWh)
1-0:2.8.1(008662.275*kWh)
1-0:2.8.2(018696.009*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.384*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00006)
1-0:52.32.0(00008)
1-0:72.32.0(00006)
1-0:32.36.0(00009)
1-0:52.36.0(00005)
1-0:72.36.0(00004)
0-0:96.13.0()
1-0:32.7.0(239.3*V)
1-0:52.7.0(234.0*V)
1-0:72.7.0(235.2*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.198*kW)
1-0:41.7.0(00.097*kW)
1-0:61.7.0(00.094*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)

### DSMR-reader version

v5.11

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script
DSMR-READER
    App | Python | Database                                                       v5.11 | v3.9.2 | postgresql
    BE sleep | DL sleep | Retention | Override                                    1.0s | 0.5s | 8784h | False
    Latest telegram version read | Parser settings                                                 "50" | "4"

DATA
    Telegrams total (est.)                                                                           28207616
    Consumption records electricity | gas (est.)                                              621717 | 109796

UNRESOLVED ISSUES
    Approximately 28207616 readings stored, consider data cleanup (if not already enabled)                                     now
    Database growing large: 8638 MB, consider data cleanup (if not already enabled)                                     now

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                8192 MB
```

### Optional: Smart meter telegram

```shell-script
1-3:0.2.8(50)
0-0:1.0.0(250510220202S)
0-0:96.1.1(4530303334303037343732373331383139)
1-0:1.8.1(010823.730*kWh)
1-0:1.8.2(012200.300*kWh)
1-0:2.8.1(008662.275*kWh)
1-0:2.8.2(018696.009*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.320*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00006)
1-0:52.32.0(00008)
1-0:72.32.0(00006)
1-0:32.36.0(00009)
1-0:52.36.0(00005)
1-0:72.36.0(00004)
0-0:96.13.0()
1-0:32.7.0(240.1*V)
1-0:52.7.0(237.8*V)
1-0:72.7.0(234.9*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(00.133*kW)
1-0:41.7.0(00.105*kW)
1-0:61.7.0(00.073*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!CE55
```

dennissiemensma, 2025-05-10:
Waarschijnlijk loop je tegen het issue aan in #2000. De suggestie van daar uitvoeren in de database fixt het mogelijk: SELECT setval(pg_get_serial_sequence('dsmr_mqtt_message', 'id'), 1, false);

cranberryNL, 2025-05-10:
Dank Dennis voor je zeer snelle reactie! Dit is inderdaad het issue waar ik ook tegen aanliep. Sorry, 'k had niet goed genoeg gezocht blijkbaar.. 
De waarde van dsmr_mqtt_message_id_seq was 2154436354. Nadat ik 'm op 1 had gezet, werkt het weer als vanouds.

Betekent dit dat ik eens in de zoveel jaar even deze waarde op 1 moet zetten? Ben je misschien voornemens om in een volgende release van de applicatie hiervoor een fix op te nemen?

dennissiemensma, 2025-05-21:
> Betekent dit dat ik eens in de zoveel jaar even deze waarde op 1 moet zetten? 
Ja dat is helaas nu wel de fix. Het hangt er erg van af hoeveel MQTT berichten je verbruikt. 
In principe ben je dan weer twee miljard berichten verder: `2.154.436.354`

> Ben je misschien voornemens om in een volgende release van de applicatie hiervoor een fix op te nemen?
Voor nu niet. De echte fix is wisselen naar UUIDs, maar dat heeft heel veel voeten in de aarde.

closed 2025-05-21 by dennissiemensma