dsmrreader/dsmr-reader #2070: Uitsplitsing L1 / L2 / L3 ook in realtime getallen, niet enkel in grafiek

gschot, 2025-06-25:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Feature or idea

### Description

Ik gebruik dsmr reader al een aantal jaar met volle tevredenheid in een docker container, enorm tevreden over de mogelijkheden. Enige tijd geleden heb ik echter mijn aansluiting laten verzwaren van 1x25A naar 3x25A vanwege overstappen naar o.a. electrisch koken ipv gas. Hierdoor is het totaalverbruik nu opgedeeld in verbruik op L1, L2 en L3. Als ik in de live grafieken kijk, is hier een cumulatief geheel over te vinden waarbij ook individuele fasen gekozen kunnen worden, mooi! Ik zou voor realtime inzicht ook graag in de eerste group bij Dashboard en ook bij Live grafieken niet enkel het totaal verbruik willen zien, maar ook uitgesplitst per fase. 

### DSMR-reader version

v5.11

### DSMR-reader platform

Docker (e.g. Xirixiz's DSMR-reader Docker)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-06-26:
Bedankt voor je verzoek. Ik denk dat ik niet helemaal begrijp wat je precies bedoelt.

Op het Dashboard staan alleen kWh-totalen en de slimme meters houden dat helaas niet bij per fase (alleen de Wattage van "een" bepaald moment, wat je in de Live grafieken ziet)

gschot, 2025-06-26:
Dank voor je snelle reactie. Ik bedoelde in de groep 'Nu', waar op dit moment enkel het realtime totale verbruik wordt getoond (L1 + L2 + L3). Ik zou het handig vinden als daar ook het realtime verbruik van de 3 fasen afzonderlijk wordt getoond. Dit zou dan bijvoorbeeld ook bovenaan de live grafieken kunnen worden getoond. De individuele actuele verbruikswaarden zijn aanwezig in Pphase currently delivered l1/2/3

De reden voor de feature request was dat bij hoger verbruik, het al wat meer inzicht geeft welk deel van de groepen dit veroorzaken.

dennissiemensma, 2025-06-26:
Bedankt voor je aanvulling. Het lastige daar aan is dat iedere gebruiker weer een andere voorkeur heeft om wat waar te tonen en ik wil voorkomen dat er tientallen opties komen voor elke mogelijke combinatie.

Wellicht dat de grafieken ooit op de kop gaan, waardoor er meer te zien is en wat ook makkelijker in of uit te schakelen is in de grafieken zelf.

gschot, 2025-06-26:
Het leek mij een nuttige aanvulling, maar ik snap je beredenering en respecteer je insteek.

closed 2025-07-07 by dennissiemensma