dsmrreader/dsmr-reader #2078: Status of the project?

bertusdebruin, 2025-08-08:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Aangezien de voertaal in Nederlands is aangegeven, heel simpel de volgende vraag: Ik ga op korte termijn verhuizen naar een nieuwbouwwoning. Aangezien 'dsmr-reader', bijna 1.5 jaar niet is ge-updated? Als oprechte vraag, heeft het nu nog zin om een dsmr-reader systeem op een Raspberry Pi compu te realiseren? Mocht dsmr-reader gezien de geringe updates, komen te vervallen? Graag even een berichtje terug, dan kan ik e.v. andere opties overwegen. Vriendelijke groet, Bertus.

### DSMR-reader version

The latest vserion.

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-08-08:
Hey Bertus, DSMR-reader vervalt niet, maar zal geen grote features meer krijgen.

De focus zit op het definitief switchen naar container-hosting, zodat systeem-updates en herinstallaties vrij simpel zijn. Voor de rest zit er nog een bug (#1770) in die daarna opgelost wordt. De rest hangt erg af van het animo onder gebruikers.

----

DSMR-reader is inmiddels bijna 10 oud en komt uit een tijd waar er nog amper (of relatief weinig) mogelijkheden waren om live je verbruik te zien. Inmiddels ondersteunt vrijwel elke energieleverancier die feature en is de wereld ook verder qua slim auto laden etc etc.
Daarnaast zijn er ook aanbieders waarbij je betaalt voor de opslag in de cloud en thuis alleen een dongle in de slimme meter hoeft te duwen.

Het hangt dus erg van je eigen situatie af wat je wil of nodig hebt...

dennissiemensma, 2025-08-24:
De wisseling naar containers is vooral vanwege de hoeveelheid tijd die het kost om alle gebruikersspecifieke issues op te lossen die gerelateerd zijn aan wijzigingen in het OS, outdated Python, een fout halvewege de installatie, etc. Net als het ondersteunen van meerdere Python-versies voor meerdere OS'en, waarbij dependencies soms weer niet alles ondersteunen, etc etc.

Ik was er eerder ook wat terughoudend in, maar achteraf heeft het me heel veel tijd gekost die ik anders in DMSM-reader zelf had kunnen steken, wat tevens ook de spagaat is in de laatste jaren qua ondersteuning vs ontwikkeling. Maar dat is achteraf altijd makkelijk praten uiteraard. 
De migratie naar containers gaat ook wel wat tijd kosten om uit te werken, maar hopelijk is het daarna voor iedereen makkelijker om DSMR-reader bij te werken zonder extra werk.

Het staat je verder vrij om geen containers te gebruiken, maar dan zul je je zelf wat moeten verdiepen in het installeren van alle requirements en alles.

bartchampagne, 2025-09-18:
Als fan van dsmrreader hoop ik dat het project maintained gaat blijven, nieuwe features zijn voor mij niet echt nodig.
Helaas teert de huidige code hier en daar op legacy dependancies (django <4 en pytz bv) en heb ik niet genoeg Python skills om dat zelf bij te werken.
Recent heb ik een procedure uitgewerkt om dsmrreader te laten werken op Debian 13 Trixie (Python3.13) met zo recent mogelijke packages - ik hoop daar komend weekend een diff voor klaar te hebben, die ik dan gerust ergens wil publiceren.

(Oa) de Sum/aggregate functie in 'dsmr_stats/services.py' teert echter op django3.2.x (die EOL is), waardoor ik django (nog) niet kan upgraden naar een hogere versie.

dennissiemensma, 2025-09-19:
@bartchampagne dank voor je reactie. Ik ben momenteel al vrij ver met de upgrade, maar daar zit nog wel wat werk in omdat het samen gaat met de switch naar containers. In de development branch staat een opzet. Momenteel ben ik echter op vakantie, dus ik kan er pas na september mee verder.

bertusdebruin, 2025-10-31:
@dennissiemensma Hoi Dennis. Wat is de status van de upgrade? Zoals ik eerder heb aangegeven, sta ik sceptisch t.o.v. een dsmr-reader op container basis. Ik wil een container setup uiteraard graag eerst testen, maar waar vind ik de documentatie en het startschot als optie?

xirixiz, 2025-11-05:
Hoi @bertusdebruin, goed om weer te lezen over een tevreden gebruiker van DSMR Reader van @dennissiemensma. Dennis kan niet genoeg complimenten krijgen voor dit mooie project 👍.

Zoals je al aangeeft: de bekende dependency hell wordt met een containeroplossing netjes opgelost. Ben je al enigszins bekend met het containerconcept? Zo niet, dan leg ik het graag even kort uit.

Ik las dat je op Debian 13 draait (goede keuze 😄). Zelf werk ik ook het liefst met Debian, al heb ik DSMR Reader in dit geval draaien binnen een container op mijn Synology NAS. Daarom zou een naam als "DSMR Reader Containerized" eigenlijk beter passen, want de oplossing is platform-onafhankelijk, daar ligt nog wat werk om te verbeteren aan mijn kant.

Wat je kunt doen als je DSMR Reader op Debian 13 draait, is Podman installeren. Dat is een open-source containerplatform, vergelijkbaar met Docker, maar dan zonder daemon en volledig rootless mogelijk. In het begin even wennen misschien, maar zodra je het door hebt, werkt het super intuitief.

Niet echt relevant, maar persoonlijk draai ik "alles" in containers:
[image]

Podman op Debian 13:
```
sudo apt-get update
sudo apt-get install podman podman-compose crun
sudo apt-get install podman-docker
```

In .basrc, of een variant daarvan (niet perse nodig, maar gemak omdat docker-compose vaak voorkomt in voorbeelden):
```
alias docker-compose='podman-compose'
```

Controlleer of Podman (en drivers) naar behoren werkt:
```
podman info --debug
```

Mogelijk komt daar nog een warning en/of error uit, deel dat dan maar even, vaak is dat een rechten issue oid, iig eenvoudig op te lossen.

Vervolgens kan je hiermee verder: https://github.com/xirixiz/dsmr-reader-docker

Als je DSMR Reader op een ander apparaat wilt draaien, gelden dezelfde basisvereisten als voor de reguliere DSMR Reader-installatie. Daarnaast moet het systeem ondersteuning bieden voor Podman en/of Docker. Een Raspberry Pi is een goede optie, maar als je al een apparaat hebt dat verbonden is met de P1-poort, dan ligt het voor de hand om dat systeem te gebruiken.

xirixiz, 2025-11-05:
Met een beetje hulp van een AI tool wat extra info 😄 

Hier is een uitleg in begrijpelijke taal over het verschil tussen het draaien van DSMR Reader rechtstreeks op Debian 13 en het draaien in een container met Podman of Docker:

🐧 DSMR Reader rechtstreeks installeren op Debian 13

Als je DSMR Reader direct op Debian 13 installeert, dan:
- Installeer je allerlei onderdelen op je systeem, zoals Python, PostgreSQL, en specifieke Python-pakketten.
- Wordt de software onderdeel van je "normale" systeem, tussen andere applicaties.
- Moet je zelf zorgen dat alle versies van programma’s goed samenwerken (denk aan Python-versies, beveiligingsupdates, dependencies, etc.).
- Er is meer handmatig werk voor installatie, configuratie en updates.
- Als er iets misgaat, kun je in de knoop raken met systeeminstellingen of andere software (de bekende dependency hell).

👉 Dit werkt prima voor één enkele app, maar wordt lastig zodra je meerdere apps draait die afhankelijk zijn van dezelfde onderdelen, maar dan in verschillende versies.

📦 DSMR Reader draaien in een container (Podman of Docker)

Met een containerplatform zoals Podman of Docker draait DSMR Reader in een geïsoleerde omgeving:
- De volledige applicatie (inclusief alle afhankelijkheden zoals de juiste Python-versie, bibliotheken, enz.) zit ingepakt in een kant-en-klaar pakket: de container  (geen systeemvervuiling).
- Je draait die container op een soort mini-virtueel systeem, zonder dat je je Debian-installatie vol hoeft te stoppen met extra software  (platform onafhankelijk).
- Je kunt verschillende applicaties naast elkaar draaien die hun eigen versies van bijvoorbeeld Python of PostgreSQL hebben, zonder dat ze elkaar in de weg zitten (geen afhankelijkheden en/of conflictrisico's_.
- Updaten wordt super eenvoudig: je downloadt gewoon een nieuwe container image en herstart de container.
- Het werkt hetzelfde op verschillende apparaten: Debian, Ubuntu, een Raspberry Pi, een Synology NAS, etc.

👉 Dit maakt je systeem schoner, stabieler en makkelijker te beheren, vooral als je meerdere apps wilt draaien.

bartchampagne, 2025-11-05:
Eventjes inpikken for good measure :-)

Zelf draai ik nu meerdere gedecentraliseerde instances van dsmrreader, elk rechtstreeks op Debian13.
(Ik ben de ICT man van de familie dus ik 'mag' intussen al een meter of 4 uitlezen en loggen)
Een switch naar een gecentraliseerdere container versie zou dus interessant zijn en staat op m'n TODO.

In mijn geval zou ik dan graag 1 enkele Postgres instance draaien voor alle dsmrreader containers maar hoe ik dat moet regelen is nog even uitzoeken.
Momenteel ook nog geen ervaring met containers dus 't gaat nog een learning curve worden.

"Als ik wat meer tijd heb" wil ik gerust een hele hoop testen.
Eerste horde is m'n meters terug online krijgen, Fluvius (BE) heeft iets kapot gedaan en 't sleept intussen al een maand aan :(

En ja, Dennis verdient tonnen lof voor deze software en het werk dat 'ie er in steekt!

dennissiemensma, 2025-11-10:
@xirixiz bedankt voor de aanvulling! Ik ga er mee spelen en kijken of ik het zelf ook werkend krijg en het dan in een handleiding kan gieten

@bartchampagne waarschijnlijk kun je wel meerdere databases aanmaken in 1 postgres instantie. En als dat niet in een container kan, kun je altijd postgres nog op je host installeren. De DB draaien in een container is vooral een nice toe have, maar de echte winst voor containers bij DSMR zit in de container van DSMR-reader zelf, daar hoef je dan uiteindelijk amper nog wat voor te installeren.

xirixiz, 2025-11-11:
Voor container oplossingen geldt over het algemeen zoveel mogelijk scheiden, maar persoonlijk vind ik dat geen toegevoegde waarde hebben voor DSMR Reader. Postgres installeren en op de host draaien is een prima oplossing inderdaad. Ik draai het in een container om verhuizen en upgrades eenvoudig te houden. Mss kunnen jullie hier iets mee.

dsmrdb start met database dsmrreader1, en als pg user dmsrreader. Initiele waarden meegeven aan de container is mandatory.
```
dsmrdb:
  image: postgres:17-alpine
  container_name: dsmrdb
  restart: always
  env_file: .env
  networks: [proxy]
  extra_hosts:
    - "nas:${HOST_IP}"
  volumes:
    - /etc/localtime:/etc/localtime:ro
    - <volume of datapad>/data/dsmr/data17:/var/lib/postgresql/data:rw
  environment:
    POSTGRES_USER: dsmrreader
    POSTGRES_PASSWORD: ${DSMR_DB_PASSWORD}
    POSTGRES_DB: dsmrreader1
```

Daarna met podman/docker de overige databases aanmaken, met dezelde pg user (indien wenselijk ook een andere owner aanmaken)
```
docker exec -it dsmrdb psql -U dsmrreader -d postgres -c "CREATE DATABASE dsmrreader2 OWNER dsmrreader;"
```

Dan container 1 (waarden kunnen afwijken uiteraard):
```
dsmr1:
  image: ghcr.io/xirixiz/dsmr-reader-docker:latest
  container_name: dsmr
  restart: always
  depends_on:
    - dsmrdb
  networks: [proxy]
  ports:
    - "7777:80"
    - "7778:443"
  cap_add:
    - NET_ADMIN
  devices:
    - /dev/ttyUSB0:/dev/ttyUSB0
  env_file: .env
  environment:
    TZ: Europe/Amsterdam
    DJANGO_TIME_ZONE: Europe/Amsterdam    
    DJANGO_DATABASE_HOST: dsmrdb
    DJANGO_DATABASE_PORT: "5432"
    DJANGO_DATABASE_NAME: dsmrreader1
    DJANGO_DATABASE_USER: dsmrreader
    DJANGO_DATABASE_PASSWORD: ${DSMR_DB_PASSWORD}
    DSMRREADER_ADMIN_USER: ${DSMR_ADMIN_USER_1}
    DSMRREADER_ADMIN_PASSWORD: ${DSMR_ADMIN_PASSWORD_1}
    VACUUM_DB_ON_STARTUP: "true"
  volumes:
    - /etc/localtime:/etc/localtime:ro
```

Dan container 2 (waarden kunnen afwijken uiteraard):
```
dsmr2:
  image: ghcr.io/xirixiz/dsmr-reader-docker:latest
  container_name: dsmr
  restart: always
  depends_on:
    - dsmrdb
  networks: [proxy]
  ports:
    - "7777:80"
    - "7778:443"
  cap_add:
    - NET_ADMIN
  devices:
    - /dev/ttyUSB1:/dev/ttyUSB1
  env_file: .env
  environment:
    TZ: Europe/Amsterdam
    DJANGO_TIME_ZONE: Europe/Amsterdam    
    DJANGO_DATABASE_HOST: dsmrdb
    DJANGO_DATABASE_PORT: "5432"
    DJANGO_DATABASE_NAME: dsmrreader2
    DJANGO_DATABASE_USER: dsmrreader
    DJANGO_DATABASE_PASSWORD: ${DSMR_DB_PASSWORD}
    DSMRREADER_ADMIN_USER: ${DSMR_ADMIN_USER_2}
    DSMRREADER_ADMIN_PASSWORD: ${DSMR_ADMIN_PASSWORD_2}
    VACUUM_DB_ON_STARTUP: "true"
  volumes:
    - /etc/localtime:/etc/localtime:ro
```