dsmrreader/dsmr-reader #2079: Hoe gehele backup te maken van een DSMR installatie en deze te restoren in een docker?

woutervanelten, 2025-08-24:
### Language / Voertaal

🇳🇱 Nederlandstalig

### Help yourself

- [x] I've checked [the how-to guides](https://dsmr-reader.readthedocs.io/nl/v5/how-to/index.html)
- [x] I've checked previous [questions](https://github.com/dsmrreader/dsmr-reader/discussions)
- [x] I've checked (recent) [previous issues](https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+is%3Aclosed+sort%3Aupdated-desc) and their solutions
- [x] I've checked the [logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html) *(when applicable)*

### Inquiry

Help with installation or upgrade

### Description

Ik heb DSMR al een hele tijd draaien in een LXC.
Nu wil ik een schone install beginnen op m'n nieuwe server in een podman quadlet.

Hoe kan ik alle data van de oude bewaren en daarna importeren in de nieuwe omgeving?

### DSMR-reader version

v5.10.4

### DSMR-reader platform

Native (e.g. manual installation)

### Optional: Debug info dump (of DSMR-reader)

```shell-script

```

### Optional: Smart meter telegram

```shell-script

```

dennissiemensma, 2025-08-24:
DSMR-reader bestaat alleen uit je data in de database, als je daar een export van maakt dan kun je die gebruiken in dezelfde of een nieuwere versie van DSMR-reader.

Het is belangrijk dat je een lege database maakt in PostgreSQL en de nieuwe installatie van DSMR-reader **nog niet start**, want anders is er een kans dat de database gevuld wordt met een lege tabelstructuur en die wil je uit je oude database hebben.

-----

Export (duurt even afhankelijk van je data):
```
sudo su - dsmr
./manage.py dsmr_backup_create --full

# "Created full backup: /home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-backup-Sunday.sql.gz"
```

----

Kopieer dat bestand naar je nieuwe databaseserver

```
# In je DB container
psql -U dsmrreader -f dsmrreader-postgresql-backup-Sunday.sql.gz dsmrreader
```

----

Als je daarna DSMR-reader start, zou die eventuele updates moeten doorvoeren en daarna moeten werken met je data.

Je kunt dit meerdere keren proberen totdat je tevreden bent over het resultaat en daarna permanent wisselen. Laat maar weten als het niet lukt.

woutervanelten, 2025-08-24:
De admin interface kan ik inderdaad wel bij.
Maar met die dropbox sync (die al gelukt is) heb ik alle data? (Info vanaf 2016 in mijn geval)
En welk van die database files (want krijg er meerdere) moet ik dan inladen in dsmr?
Of is er ook een import functie in dsmr?

dennissiemensma, 2025-08-25:
Fijn dat het gelukt is! Er is geen import-functie, omdat DSMR-reader niet zonder database kan draaien (en je dat voor de import juist wel weer wilt).

Waar liep je uiteindelijk nog tegenaan? Of was het vooral uitzoeken hoe je het allemaal in de nieuwe DB-container kreeg?